microsoft/AirSim #4589: How to get equal exposure for all camera for stiching

JiancongWang, 2022-06-23:
## Question
### What's your question?

I am running 5 virtual cameras with fov 90 degree in +-xyz (omit the -y for now since it points to the ground) direction to form a cube map and I want to stich them together to form an epirectangular image. But I found that the exposure of the images are drastically different with each other even when the image are collected by python script simultaneouly (so there should not be any effect of light direction change). This is from the landscapeMountain scene.  I suppose this is due to auto exposure. So I want to ask if there is a way to set manual exposure level here? 

[image]

Here is how I added the virtual camera in the settings.json.
"Vehicles": {
    "PhysXCar": {
      "VehicleType": "ComputerVision",
      "DefaultVehicleState": "",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "RC": {
        "RemoteControlID": -1
      },
      "Cameras": {   
        "MyCamera0": {
          "X": 2, "Y": 0, "Z": -1,
		  "Pitch": 0, "Roll": 0, "Yaw": -90
        },
        "MyCamera1": {
          "X": 2, "Y": 0, "Z": -1,
		  "Pitch": 0, "Roll": 0, "Yaw": 0
        },
	"MyCamera2": {
          "X": 2, "Y": 0, "Z": -1,
		  "Pitch": 0, "Roll": 0, "Yaw": 90
        },
	"MyCamera3": {
          "X": 2, "Y": 0, "Z": -1,
		  "Pitch": 0, "Roll": 0, "Yaw": 180
        },
	"MyCamera4": {
          "X": 2, "Y": 0, "Z": -1,
		  "Pitch": 90, "Roll": 0, "Yaw": 0
        }
      }
    }
  }

The collection script (modded from fov_change.py so the rest of it is the same):

requests = [airsim.ImageRequest("MyCamera0", airsim.ImageType.Scene),
            airsim.ImageRequest("MyCamera1", airsim.ImageType.Scene),
            airsim.ImageRequest("MyCamera2", airsim.ImageType.Scene),
            airsim.ImageRequest("MyCamera3", airsim.ImageType.Scene),
            airsim.ImageRequest("MyCamera4", airsim.ImageType.Scene)
            ]
client.simSetCameraFov("MyCamera0", 90)
client.simSetCameraFov("MyCamera1", 90)
client.simSetCameraFov("MyCamera2", 90)
client.simSetCameraFov("MyCamera3", 90)
client.simSetCameraFov("MyCamera4", 90)

responses = client.simGetImages(requests)
save_images(responses, "new_fov_90")

### Include context on what you are trying to achieve

#### Context details

This is running UR 4.27.2 with the latest airsim from github.

jonyMarino, 2022-06-24:
Hi @JiancongWang! This may be due to missing image postprocessing. Can you try applying the solution given in [this](https://github.com/microsoft/AirSim/issues/387#issuecomment-932607229) thread?
I like that image. How did you generate it?

JiancongWang, 2022-06-24:
I think that post processing is not the problem here. In real camera terms, I want to make sure the camera ISO/aperature/exposure time are all the same for the image so that their exposure is as closed as possible. I think now in airsim the image is auto adjusted to fit a certain target. I want to turn that off and instead manually fixed the picture intensity.   
I tried histogram match all the image to the middle one. It certainly looks better but the seam between stiching is still very bad. 

The stiching is done by the py360 convert package. So once you have the 5 fov90 views by the 5 cameras in the settings.json, you can treat it as the cube map faces and convert that to equirectangular using this. 
https://github.com/sunset1995/py360convert

jonyMarino, 2022-06-24:
Nice package, thanks!

#3949 changed the default unreal camera to the cinematographic one. I think what you are searching for is [this API](https://microsoft.github.io/AirSim/api_docs/html/#airsim.client.VehicleClient.simSetFocusAperture) method added by that PR

JiancongWang, 2022-06-27:
Thank you for sending me this! I will give it a try.