microsoft/AirSim #4593: Error building ros2 package

fxa76, 2022-06-30:
## Question
### What's your question?
When building Ros2 package I get the following error, what am I doing wrong ? 
/home/fxa/AirSim/ros2/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp: In member function ‘sensor_msgs::msg::PointCloud2 AirsimROSWrapper::get_lidar_msg_from_airsim(const msr::airlib::LidarData&, const string&, const string&) const’:
/home/fxa/AirSim/ros2/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp:685:138: error: ‘from_nanoseconds’ is not a member of ‘rclcpp::Duration’
  685 |                 auto transformStampedENU = tf_buffer_->lookupTransform(AIRSIM_FRAME_ID, vehicle_name, rclcpp::Time(0), rclcpp::Duration::from_nanoseconds(1));

### Include context on what you are trying to achieve
I am trying to build the Ros2 package using this command: 
colcon build --cmake-args -DCMAKE_C_COMPILER=gcc-8 --cmake-args -DCMAKE_CXX_COMPILER=g++-8 --cmake-args -DCMAKE_BUILD_TYPE=Debug

#### Context details
OS: Ubuntu 20.04 in wsl 2 on windows 10.
AirSim version: ~/AirSim$ git checkout
Your branch is up to date with 'origin/master'
Ros Distro:  printenv ROS_DISTRO foxy
https://gist.github.com/fxa76/cbb477d3946f882c92f5d747166804ee

### Include details of what you already did to find answers

alonfaraj, 2022-06-30:
Hi @fxa76,
After this [PR](https://github.com/microsoft/AirSim/pull/4532), ros2 package support only galactic distro.
You can revert those changes or upgrade to galactic.

fxa76, 2022-06-30:
Thanks it builds and runs now.

closed 2022-06-30 by fxa76