microsoft/AirSim #4595: XCode Build fails on MacOS

AlexGisi, 2022-06-30:
## Bug report 

- AirSim Version/#commit: [4b758d0](https://github.com/microsoft/AirSim/commit/4b758d00907855e65fd9dca9483e3c284f2210dd)
- UE/Unity version: 4.27.2
- autopilot version: NA
- OS Version: Mac 10.15
- XCode version: 12.3

### What's the issue you encountered?

After running the setup and build script for AirSim, the Xcode build step of the [setup](https://microsoft.github.io/AirSim/build_macos/) fails with two errors:

`ERROR: Couldn't find target rules file for target 'ShaderCompileWorker' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.`

`ERROR: Unhandled exception: System.UnauthorizedAccessException: Access to the path "/Users/alex/.../AirSim/Unreal/Environments/Blocks/Intermediate/Build/SourceFileCache.bin" is denied.`

The permission denied error is confusing because the file AirSim/Unreal/Environments/Blocks/Intermediate/Build/SourceFileCache.bin doesn't seem to exist.

The setup, build and generateProjectFiles scripts succeed without errors.

### How can the issue be reproduced?

```
cd Airsim/
./setup.sh
sudo ./build.sh
cd Unreal/Environments/Blocks
sudo ./GenerateProjectFiles.sh "/Users/Shared/Epic Games/UE_4.27"
```
Then open the xcode project and hit the run button.

### Include full error message in text form
```

Showing Recent Messages
Found mono via known Mono.framework path

Running system mono/msbuild, version: Mono JIT compiler version 6.12.0.93 (2020-02/620cf538206 Tue Aug 25 14:04:52 EDT 2020)

Processing build for Target=BlocksEditor Platform=Mac Configuration=Development -architecture=x86_64 /Users/alex/.../AirSim/Unreal/Environments/Blocks/Blocks.uproject 

Creating makefile for ShaderCompileWorker (no existing makefile)

ERROR: Couldn't find target rules file for target 'ShaderCompileWorker' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.

       Location: /Users/Shared/Epic Games/UE_4.27/Engine/Intermediate/Build/BuildRules/UE4Rules.dll

       Target rules found:

Running Engine/Binaries/DotNET/UnrealBuildTool.exe BlocksEditor Mac Development /Users/alex/.../AirSim/Unreal/Environments/Blocks/Blocks.uproject -architecture=x86_64

Creating makefile for BlocksEditor (no existing makefile)

Parsing headers for BlocksEditor

ERROR: Unhandled exception: System.UnauthorizedAccessException: Access to the path "/Users/alex/.../AirSim/Unreal/Environments/Blocks/Intermediate/Build/SourceFileCache.bin" is denied.

         at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x0019e] in <3f2977eb306b45388f224fc6cc4db2c4>:0 

         at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) [0x00000] in <3f2977eb306b45388f224fc6cc4db2c4>:0 

         at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare)

         at System.IO.File.Open (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) [0x00000] in <3f2977eb306b45388f224fc6cc4db2c4>:0 

         at UnrealBuildTool.SourceFileMetadataCache.Write () [0x00010] in <ee25ef7ab1534d21a45b7cf37b6e3a3f>:0 

         at UnrealBuildTool.SourceFileMetadataCache+<>c.<SaveAll>b__20_0 (UnrealBuildTool.SourceFileMetadataCache Cache) [0x00008] in <ee25ef7ab1534d21a45b7cf37b6e3a3f>:0 

         at System.Threading.Tasks.Parallel+<>c__DisplayClass44_0`2[TSource,TLocal].<PartitionerForEachWorker>b__1 (System.Collections.IEnumerator& partitionState, System.Int32 timeout, System.Boolean& replicationDelegateYieldedBeforeCompletion) [0x00103] in <3f2977eb306b45388f224fc6cc4db2c4>:0 

       --- End of stack trace from previous location where exception was thrown ---

       

         at System.Threading.Tasks.Parallel+<>c__DisplayClass44_0`2[TSource,TLocal].<PartitionerForEachWorker>b__1 (System.Collections.IEnumerator& partitionState, System.Int32 timeout, System.Boolean& replicationDelegateYieldedBeforeCompletion) [0x0025f] in <3f2977eb306b45388f224fc6cc4db2c4>:0 

         at System.Threading.Tasks.TaskReplicator+Replica`1[TState].ExecuteAction (System.Boolean& yieldedBeforeCompletion) [0x00000] in <3f2977eb306b45388f224fc6cc4db2c4>:0 

         at System.Threading.Tasks.TaskReplicator+Replica.Execute () [0x00023] in <3f2977eb306b45388f224fc6cc4db2c4>:0

Command ExternalBuildToolExecution failed with a nonzero exit code
```

**What's better than filing an issue? Filing a pull request :).**

AlexGisi, 2022-07-04:
I got it working after following the [Blocks environment setup](https://microsoft.github.io/AirSim/unreal_blocks/) instructions between building AirSim and running the Xcode project.

closed 2022-07-04 by AlexGisi

santurini, 2023-07-12:
Were you able to start it on MacOS? I have troubles in building the xcode project, can you help me?

AlexGisi, 2023-07-12:
> Were you able to start it on MacOS? I have troubles in building the xcode project, can you help me?

Did you follow the Blocks environment setup instructions for Linux?

https://microsoft.github.io/AirSim/unreal_blocks/

santurini, 2023-07-13:
I have folllowed [this](https://microsoft.github.io/AirSim/build_macos/) instructions. In my UnrealEngine folder I only have a Launcher subfolder.

I was still able to start the Unreal Engine Editor and opening the Blocks.uproject, skipping the Xcode building part.
But when I enter the editor I get not prompted with building AirSim. This is what I see:
[image]

Should I click the "Build" button and then "Play"?

AlexGisi, 2023-07-13:
As far as I know you have to follow the Blocks setup instructions as well. I'm not very familiar with airsim or unreal engine so I can't help you much beyond that.