microsoft/AirSim #4601: AirSim/PX4 crashes when using non-default environment from settings.json

talha-opteran, 2022-07-04:
## Bug report 

- AirSim Version/#commit: v1.7.0
- UE/Unity version: Unreal 4.27.2
- autopilot version: PX4 v.1.13 or v.1.11
- OS Version: Ubuntu 20.04

### What's the issue you encountered?

AirSim crashes when trying to load an environment other than the default when using PX4. I believe this is because a MavLink connection is created on launch for the default environment and then destroyed and recreated when a new environment is loaded. This is both in the editor and packaged version.

When launching from the editor with the new environment already loaded, AirSim seems to work fine.

### Settings

Default PX4 settings:

```
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "ClockType": "SteppableClock",
    "Default Environment": "<Any non-blocks map>"
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "LockStep": true,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPortLocal": 14540,
            "ControlPortRemote": 14580,
            "Sensors":{
                "Barometer":{
                    "SensorType": 1,
                    "Enabled": true,
                    "PressureFactorSigma": 0.0001825
                }
            },
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "COM_OBL_ACT": 1,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165
            }
        }
    }
}
```

### How can the issue be reproduced?

1. Launch PX4: `make px4_sitl_default none_iris`
2. From the editor, launch AirSim using above settings
3. The new environment will load but there will be no MavLinkConnection.
4. Open the new environment in the editor, rerun step 1 & 2, and everything should work as normal.

### Include full error message in text form
#### PX4
```
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
INFO  [parameters] BSON document size 869 bytes, decoded 869 bytes (INT32:22, FLOAT:20)
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
```
#### AirSim
```
4.27.2-0+++UE4+Release-4.27 522 0
Disabling core dumps.
Looking for binary: ../../../Config/BinaryConfig.ini
Opening mavlink connection
Disconnecting mavlink vehicle
Waiting for mavlink vehicle...
Disconnecting mavlink vehicle
Waiting for TCP connection on port 4560, local IP 127.0.0.1
Connected to SITL over TCP.
Connecting to PX4 Control UDP port 14540, local IP 127.0.0.1, remote IP 127.0.0.1 ...
Ground control connected over UDP.
received first heartbeat
PX4
Disconnecting mavlink vehicle
Opening mavlink connection
Disconnecting mavlink vehicle
Waiting for mavlink vehicle...
Disconnecting mavlink vehicle
Waiting for TCP connection on port 4560, local IP 127.0.0.1
libc++abi: terminating with uncaught exception of type std::runtime_error: TcpClientPort socket bind failed with error: 98

Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Failed to find symbol file, expected location:
"/media/Data/unreal_projects/Airsim/LinuxNoEditor/Airsim/Binaries/Linux/Airsim-Linux-Shipping.sym"
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
```

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2022-07-08:
@talha-opteran Maybe you have a firewall preventing the connection Can you check?

closed 2022-07-17 by jonyMarino

fengtudou, 2024-04-01:
Could you please confirm if this issue has been resolved?Thanks

PR #4599 (2022-07-17): Fix load level for UE editor

Fixes: #4551
Fixes: #4601

## About

Current method to load level may work for a packaged binary but does not work in the UE editor where it will infinitely load the requested level. This PR adds a check to make sure the current level is not reloaded.

## How Has This Been Tested?

1. Load project in editor.
2. Add a "Default Environment" value in settings.json
3. Launch AirSim.
4. Before fix, editor freezes due to infinitely reloading requested level. After fix, new level is loaded.

## Screenshots (if appropriate):