microsoft/AirSim #4617: AirSim RVIZ Connection Errors

EvelynM7, 2022-07-12:
## Bug report 

- AirSim Version/#commit: Clone from https://github.com/Microsoft/AirSim.git as of yesterday
- UE/Unity version: Unreal Engine 4.27.2
- autopilot version: N/A
- OS Version: Windows 10 Pro, 64 bit, 19044.1766
- Other: Ubuntu 20.04, WSL 2, ROS2 galactic

### What's the issue you encountered?

I have now tried many processes of setting up Ubuntu, ROS, RVIZ, and AirSim as well as different edits to the airsim_ros_wrapper code. The following steps are for the setup that has worked best for me, but I am still running into at least one error getting RVIZ to connect properly with AirSim showing the camera.

As can be seen in the images attached below, this gave me Status: Warn like #4509 The error for Camera Info says "No CameraInfo received on [/airsim_node/Drone1/Test0_0/camera_info]. Topic may not exist."

Both @zimmy87 and @alonfaraj also showed a screenshot with Frames in #4509. I only have world_enu and world_ned as options in RVIZ. I think this and the lack of topics could be causing the error. How do I fix both of these? I've included the RVIZ screenshot and the list of topics from the terminal (there are some missing, I think, when compared to @alonfaraj's list).

Any assistance would be greatly appreciated. Thank you!

[image]
[image]

### Settings

```
{
 "SimMode" : "Multirotor", 
 "SettingsVersion" : 1.2, 
 "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "Vehicles": {
    "Drone1": {
        "VehicleType": "SimpleFlight",
        "AutoCreate": true,
        "Sensors": {
            "Barometer":{
                "SensorType": 1,
                "Enabled": true,
                "PressureFactorSigma": 0.0001825
            },
            "MyLidar1": {
                "SensorType": 6,
                "Enabled" : true,
                "NumberOfChannels": 16,
                "PointsPerSecond": 10000,
                "X": 0, "Y": 0, "Z": -1,
                "DrawDebugPoints": true
            },
            "MyLidar2": {
                "SensorType": 6,
                "Enabled" : true,
                "NumberOfChannels": 4,
                "PointsPerSecond": 10000,
                "X": 0, "Y": 0, "Z": -1,
                "DrawDebugPoints": true
            }
        },
        "Cameras": {
            "Test0": {
                "CaptureSettings": [
                  {
                    "ImageType": 0,
                    "Width": 256,
                    "Height": 144,
                    "FOV_Degrees": 90,
                    "AutoExposureSpeed": 100,
                    "AutoExposureBias": 0,
                    "AutoExposureMaxBrightness": 0.64,
                    "AutoExposureMinBrightness": 0.03,
                    "MotionBlurAmount": 0,
                    "TargetGamma": 1.0,
                    "ProjectionMode": "",
                    "OrthoWidth": 5.12
                  }
                ],
                "NoiseSettings": [
                  {
                    "Enabled": false,
                    "ImageType": 0,

                    "RandContrib": 0.2,
                    "RandSpeed": 100000.0,
                    "RandSize": 500.0,
                    "RandDensity": 2,

                    "HorzWaveContrib":0.03,
                    "HorzWaveStrength": 0.08,
                    "HorzWaveVertSize": 1.0,
                    "HorzWaveScreenSize": 1.0,

                    "HorzNoiseLinesContrib": 1.0,
                    "HorzNoiseLinesDensityY": 0.01,
                    "HorzNoiseLinesDensityXY": 0.5,

                    "HorzDistortionContrib": 1.0,
                    "HorzDistortionStrength": 0.002
                  }
                ],
                "Gimbal": {
                  "Stabilization": 0,
                  "Pitch": 0, "Roll": 0, "Yaw": 0
                },
                "X": 0, "Y": 0, "Z": 0,
                "Pitch": 0, "Roll": 0, "Yaw": 0
            }
        }
    }
}
}
```

### How can the issue be reproduced?

#### ROS2 Setup from scratch: (https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html#install-ros-2-packages)
```
# In Windows Command Prompt
($ wsl --install)
($ wsl -l -o)
$ wsl --install -d Ubuntu-20.04
# Follow Ubuntu terminal setup prompts
($ wsl -l -v)

# In Ubuntu:
$ apt-cache policy | grep universe
$ sudo apt install software-properties-common
$ sudo add-apt-repository universe

$ sudo apt update && sudo apt install curl gnupg lsb-release
$ sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
$ echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(source /etc/os-release && echo $UBUNTU_CODENAME) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null

$ sudo apt update
$ sudo apt upgrade
$ sudo apt install ros-galactic-desktop
$ source /opt/ros/galactic/setup.bash
$ echo "source /opt/ros/galactic/setup.bash" >> ~/.bashrc
$ source ~/.bashrc

$ sudo apt-get install ros-galactic-tf2-sensor-msgs ros-galactic-tf2-geometry-msgs ros-galactic-mavros*
$ sudo apt install python3-pip
$ sudo apt install python3-rosdep python3-rosinstall python3-rosinstall-generator python3-wstool build-essential
$ sudo apt install python3-rosdep
$ sudo rosdep init
$ rosdep update
$ sudo apt install python3-colcon-common-extensions
($ pip install "git+https://github.com/catkin/catkin_tools.git#egg=catkin_tools")

$ git clone https://github.com/Microsoft/AirSim.git
$ cd AirSim
$ ./setup.sh
$ ./build.sh
($ echo "source /opt/ros/galactic/setup.bash" >> ~/.bashrc)
($ source ~/.bashrc)

# Edit files in new terminal before build or rebuild before running rviz (vi ros2/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp)
# Follow edits listed in {https://github.com/microsoft/AirSim/pull/4509/files#}

$cd ros2
$ colcon build
```

#### Launching RVIZ:
```
# Open ubuntu terminal
$ cd AirSim/ros2
$ source install/setup.bash
($ source /opt/ros/galactic/setup.bash)
$ ros2 launch airsim_ros_pkgs airsim_node.launch.py

# Open 2nd ubuntu terminal
$ cd AirSim/ros2
$ source install/setup.bash
($ source /opt/ros/galactic/setup.bash)
$ export DISPLAY=$(cat /etc/resolv.conf | grep nameserver | awk '{print $2}'):0
$ ros2 launch airsim_ros_pkgs rviz.launch.py
```

#### RVIZ Steps/Edits:
```
# Click Add
# Add Camera
# Edit Camera Topic to /airsim_node/Drone1/Test0_0/image
# Click Reset (no image or camera showed even before reset though no error code was given)
```

### Include full error message in text form
In RVIZ: "Status: Warn". The error for Camera Info also says "No CameraInfo received on [/airsim_node/Drone1/Test0_0/camera_info]. Topic may not exist."

alonfaraj, 2022-07-13:
Hi @EvelynM7,
It's hard to say what is wrong, but what I do see is that you might modified the files a bit differently than https://github.com/microsoft/AirSim/pull/4509, otherwise your topic would be `/airsim_node/Drone1/Test0_Scene/image` and not `/airsim_node/Drone1/Test0_0/image`.
It seems like you modifed the files manullay, I suggest you using https://github.com/microsoft/AirSim/pull/4509 as is, with `gh pr checkout 4509` and report back.

EvelynM7, 2022-07-13:
Hello @alonfaraj ,

I'm not sure what you mean by `gh pr checkout 4509`. I tried that command both in Git Bash and Ubuntu, but it didn't work. Could you elaborate, please?

I did go back and copy directly the changes made in #4509 as listed in {https://github.com/microsoft/AirSim/pull/4509/files#}. I also closely checked the spacing, spelling, etc., and mine is now identical to the one listed there. I am still having to manually input the topic for the camera as shown in the images attached below. I think this is where you are saying my topic should be different. I believe I followed what was written out as `Test0_0,` but `Test0_Scene` is what shows up in your topic list so I've included screenshots of that inputted topic. You can see that I still don't have the frames (even when all are enabled) and topics you had in #4509 as well. Why could this be, and how do I fix it? Are you using a different rviz file? Am I missing a step?

[image]
[image]
[image]
[image]

alonfaraj, 2022-07-14:
`gh` is github cli so you can use it to checkout a branch from a fork easier - https://cli.github.com/.  
I'm using a standard installation of ROS2 galactic included rviz2.  
Is `airsim_node` showing something? error/warning?  

You topic list doesn't make sense, make a test:
1. Kill rviz
2. Kill all other running nodes 
3. Make sure you topic list is empty (except `/parameter_events` and `/rosout`)
4. Run `airsim_node`
5. Check your topic list, it should be exactly like this (I'm using your settings.json):

[image]

If not - it seems like your node crashed or didn't connect to the simulation.

EvelynM7, 2022-07-14:
There's an info comment when running `airsim_node`: `[static_transform_publisher-1] [INFO] [1657814690.091340820] [ned_to_enu_pub]: Spinning until killed publishing transform from 'world_ned' to 'world_enu'`.

Following your steps gives me the below screenshot (the last topic command that also includes `/tf_static [tf2_msgs/msg/TFMessage]` is the one after starting `airsim_node`).

[image]

alonfaraj, 2022-07-14:
Do you run the simulation on the same machine?

EvelynM7, 2022-07-14:
Yes, we do.
Do you have any suggestions on how to fix this or test why this might be happening?

alonfaraj, 2022-07-15:
You should specify the host ip which run the simulation. 

`ros2 launch airsim_ros_pkgs airsim_node.launch.py host:=<your_simulation_host_ip>`

EvelynM7, 2022-07-15:
Is the simulation host ip different then the computer IP address?

alonfaraj, 2022-07-15:
No. It's the ip address of your computer running the simulation

EvelynM7, 2022-07-18:
Thank you very much! That did seem to connect properly.

I do have a couple more questions:
- Would you share your settings.json file that gave such a clear camera view? Is something else causing mine to be fainter (see screenshot below), and how can I solve that issue?
- My topic list seems to be missing `compressed`, `compressedDepth`, and `theora` (though this looks like some sort of plugin or package). Should I have these as well or is it fine that they are not showing up for the topic command?

[image]
[image]

alonfaraj, 2022-07-19:
1. You have set x,y,z,yaw,pitch,roll in your settings and override the default so you basically placed your camera in the middle of the drone and you see 'through' it. 
Just delete those settings and use the default. Or modify them for your own needs. 
2. Yep you should install ros-galactic-image-transport-plugins to get those topics published.

closed 2022-07-19 by EvelynM7