microsoft/AirSim #4618: How to get continuous sensor readings from the multirotor?

Mengyuliu0520, 2022-07-13:
## What feature are you suggesting?
### Overview: Can we monitor the model and get a stream of sensor readings through api?
 Maybe it is good to have an api that output sensor readings continuously, since the users do not have to care about the sychronization problems.

### Smaller Details: 

### Nature of Request:  

- Addition
- Change
- Removal
  - 

## Why would this feature be useful?

alonfaraj, 2022-07-13:
Maybe it's worth using ros for that purpose?

Mengyuliu0520, 2022-07-13:
> Maybe it's worth using ros for that purpose? I think rpclib doesn't support pub-sub which is probably the way to implement such feature for python api.

Thanks for the advice! What I'm considering is a MPC-based real-time control using AirSim, to achieve this, first I need to have real-time vehicle states and I also need to access low-level control in real time, I found there are some old issues discussed about the low-level control and I will give it a try.

closed 2022-07-14 by Mengyuliu0520