microsoft/AirSim #4637: Blocks.zip in 1.8.1 Linux release is just 22 bytes.

AlibekJ, 2022-07-23:
## Question
### What's your question?
Blocks.zip in 1.8.1 Linux release is just 22 bytes.
https://github.com/microsoft/AirSim/releases/tag/v1.8.1

### Include context on what you are trying to achieve
Trying to download Blocks environment.

#### Context details

### Include details of what you already did to find answers

AlibekJ, 2022-07-30:
Hi @jonyMarino can you please help?

jonyMarino, 2022-07-31:
@AlibekJ. Yes sure. I will try to upload a package ASAP. If you have the urgency, you can package Blocks on your own. Take a look at [the docs](https://microsoft.github.io/AirSim/build_faq/#packaging-a-binary-including-the-airsim-plugin)

jonyMarino, 2022-08-02:
@AlibekJ, can you try the new binary, please?

AlibekJ, 2022-08-03:
Blocks 1.8.1 linux works fine, @jonyMarino. 
However, Building_99.zip is still 22 bytes.

jonyMarino, 2022-08-03:
@AlibekJ I'm short of time. If you are interested I can fix this. If not I will prefer to wait for someone who needs it.

AlibekJ, 2022-08-04:
@jonyMarino, thanks for your efforts. I am interested of course and I am sure others will be interested too knowing that MS decided to stop supporting this repo. I am just hoping to download/fork everything in case if the repo gets shut down completely. 

I tried compiling the missing environments but failed to do so, hence asking for your help.