microsoft/AirSim #4648: client.simSetFocalLength Issue

EvelynM7, 2022-07-29:
## Bug report 

- AirSim Version/#commit: 7/29/2022
- UE/Unity version: 4.27
- autopilot version: N/A
- OS Version: Windows 10 Pro, 19044.1826

### What's the issue you encountered?

Originally, this problem showed up in computer vision mode. I was trying to modify camera parameters in the settings.json, but when that was not working due to cv mode, I found the client.simSetFocalLength function. However, this gave the following error: `msgpackrpc.error.RPCError: rpclib: server could not find function 'simSetFocalLength' with argument count 4.` I thought this might be an issue with cv mode, so I also tried it with multirotor settings and with simGetFocalLength. Both of these are giving similar argument errors in the debug and terminal as well as in the client.py.

I tried looking up solutions to this problem and the only reason I could find was one person saying that it might be due to client and server versions, but I checked that both of mine are up to date. (There seems to be hard coded 1s for the versions in client.py?) Any suggestions or solutions would be greatly appreciated!

[image]
[image]

### Settings

```
{
  "SimMode" : "Multirotor", 
  "SettingsVersion" : 1.2, 
  "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "ClockSpeed": 10,
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 320,
        "Height": 320,
        "FOV_Degrees": 84,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      },
      {
        "ImageType": 1,
        "Width": 1920,
        "Height": 1080,
        "FOV_Degrees": 84,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      },
      {
        "ImageType": 6,
        "Width": 320,
        "Height": 320,
        "FOV_Degrees": 84,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      },
      {
        "ImageType": 2,
        "Width": 320,
        "Height": 320,
        "FOV_Degrees": 84,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      }
  ]
},
  "Vehicles": {
     "Drone1": {
         "VehicleType": "SimpleFlight",
         "AutoCreate": true,
         "Sensors": {
             "Altitude":{
                 "SensorType": 5,
                 "Enabled": true,
                 "Yaw": 0, "Pitch": -90, "Roll": 0,
                 "MaxDistance": 30,
                 "DrawDebugPoints": true
             },
             "RangeSensor":{
                 "SensorType": 5,
                 "Enabled": true,
                 "Yaw": 0, "Pitch": -50, "Roll": 0,
                 "MaxDistance": 30,
                 "DrawDebugPoints": true
             },
             "LidarDistance": {
              "SensorType": 6,
              "Enabled" : true,
              "NumberOfChannels": 16,
              "PointsPerSecond": 10000,
              "X": 1, "Y": 0, "Z": 0,
              "VerticalFOVUpper": 1,
              "VerticalFOVLower": -1,
              "HorizontalFOVStart": -15,
              "HorizontalFOVEnd": 15,
              "DataFrame": "SensorLocalFrame",
              "DrawDebugPoints": false
             },
             "Barometer":{
                 "SensorType": 1,
                 "Enabled": true,
                 "PressureFactorSigma": 0.0001825
             },
             "MyLidar1": {
                 "SensorType": 6,
                 "Enabled" : true,
                 "NumberOfChannels": 16,
                 "PointsPerSecond": 10000,
                 "X": 0, "Y": 0, "Z": -1,
                 "DrawDebugPoints": false
             },
             "MyLidar2": {
                 "SensorType": 6,
                 "Enabled" : true,
                 "NumberOfChannels": 4,
                 "PointsPerSecond": 10000,
                 "X": 0, "Y": 0, "Z": -1,
                 "DrawDebugPoints": false
             }
         },
         "Cameras": {
             "Test0": {
                 "CaptureSettings": [
                   {
                     "ImageType": 0,
                     "Width": 640,
                     "Height": 512,
                     "FOV_Degrees": 90,
                     "AutoExposureSpeed": 100,
                     "AutoExposureBias": 0,
                     "AutoExposureMaxBrightness": 0.64,
                     "AutoExposureMinBrightness": 0.03,
                     "MotionBlurAmount": 0,
                     "TargetGamma": 1.0,
                     "ProjectionMode": "",
                     "OrthoWidth": 5.12
                   }
                 ],
                 "NoiseSettings": [
                   {
                     "Enabled": false,
                     "ImageType": 0,
 
                     "RandContrib": 0.2,
                     "RandSpeed": 100000.0,
                     "RandSize": 500.0,
                     "RandDensity": 2,
 
                     "HorzWaveContrib":0.03,
                     "HorzWaveStrength": 0.08,
                     "HorzWaveVertSize": 1.0,
                     "HorzWaveScreenSize": 1.0,
 
                     "HorzNoiseLinesContrib": 1.0,
                     "HorzNoiseLinesDensityY": 0.01,
                     "HorzNoiseLinesDensityXY": 0.5,
 
                     "HorzDistortionContrib": 1.0,
                     "HorzDistortionStrength": 0.002
                   }
                 ],
                 "Gimbal": {
                   "Stabilization": 0,
                   "Pitch": 0, "Roll": 0, "Yaw": 0
                 },
                 "X": 0, "Y": 0, "Z": 0,
                 "Pitch": 0, "Roll": 0, "Yaw": 0
             }
         }
     }
 }
 }
```

### How can the issue be reproduced?

1. Run client.simSetFocalLength("focal_length, "camera_name", vehicle_name="vehicle_name", external=False) in code or debug
2. Run client.simGetFocalLength("camera_name", vehicle_name="vehicle_name", external=False) in code or debug

### Include full error message in text form
`msgpackrpc.error.RPCError: rpclib: server could not find function 'simSetFocalLength' with argument count 4.` 
`msgpackrpc.error.RPCError: rpclib: server could not find function 'simGetFocalLength' with argument count 3.`
`rpclib: server could not find function 'simSetFocalLength' with argument count 4.`
`rpclib: server could not find function 'simGetFocalLength' with argument count 3.`

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2022-08-12:
@EvelynM7, which python client version are you using?

EvelynM7, 2022-08-16:
@jonyMarino, I'm using Python 3.10.6 64 bit in Visual Studio Code.

jonyMarino, 2022-08-16:
@EvelynM7 what number do you get running this:
import airsim
airsim.__version__

EvelynM7, 2022-08-16:
I'm now getting `airsim.version
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: module 'airsim' has no attribute 'version'`.

jonyMarino, 2022-08-16:
@EvelynM7, sorry it is `airsim.__version__`

EvelynM7, 2022-08-16:
@jonyMarino, It says '1.7.0'.

jonyMarino, 2022-08-16:
Sorry, now I see you are running from source. Are you running the simulation from a binary? This is typically a version mismatch between client and server.