microsoft/AirSim #4654: DefaultSensors->Barometer/Imu/Gps/...cannot be applied to corresponding sensors

dzywater, 2022-08-03:
In my settings.json has following two configs.
---
	"DefaultSensors": 
	{
		"Distance":
		{
			"SensorType": 5,
			"Enabled": true, "DrawDebugPoints": true,
			"MinDistance": 0.2,  "MaxDistance": 100
		},
		"Lidar":
		{
			"SensorType": 6,
			"Enabled": true, "DrawDebugPoints": true,
			"NumberOfChannels": 32, "Range": 60,
			"RotationsPerSecond": 10, "PointsPerSecond": 100000,
			"VerticalFOVLower": -30, "VerticalFOVUpper": 30,
			"HorizontalFOVStart": 15, "HorizontalFOVEnd": 345,
			"DataFrame": "SensorLocalFrame", "ExternalController": false
		}
	}
---
	"Vehicles":
	{
		"drone_0":
		{
			"X": 2, "Y": 4, "Z": 0, "Roll": 0, "Pitch": 0, "Yaw": 0,
			"VehicleType": "SimpleFlight",
			"Sensors":
			{
				"front_distance": { "SensorType": 5, "X": 0.5, "Y": 0, "Z": 0 },
				"top_lidar": { "SensorType": 6, "X": 0, "Y": 0, "Z": -0.24 }
			}

jonyMarino, 2022-08-04:
@dzywater. If you specify a sensor, its defaults are overwritten by AirSim defaults. You will need to set all the parameters

dzywater, 2022-08-05:
> @dzywater. If you specify a sensor, its defaults are overwritten by AirSim defaults. You will need to set all the parameters

" You will need to set all the parameters"? 

I don't what you mean?

dzywater, 2022-08-05:
For a sensor like SensorType=6 lidar, for most situations with several lidars with same hardware params, hardware params are expected to be configured in "DefaultSensors" regions and pose in "Vehicles" region.

only reset its pose in robot body, and the hardware parameters keep same.