microsoft/AirSim #4659: Cannot open blocks form UE4

Gloriabhsfer, 2022-08-09:
## Question
### What's your question?

Hi all:
I have follow the steps form the manual for UE4 and AirSim for Linux to install and build the project. However, I encounter with question while I am trying to open the blocks Uproject. My system is Ubuntu 20.04 and Unreal_Engine 2.47.2. I try convert the project when I am open the project
### The error turns out like this

```The project could not be compiled. Would you like to open it in Null Source Code Access?

Running /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project="/usr/local/bin/AirSim/AirSim-main/Unreal/Environments/Blocks/Blocks.uproject" -TargetType=Editor -Progress -NoEngineChanges -NoHotReloadFromIDE

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
/usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Binaries/Linux
Using 'git status' to determine working set for adaptive non-unity build (/usr/local/bin/Unreal_Engine/UnrealEngine-4.27).
Creating makefile for BlocksEditor (no existing makefile)
WARNING: Library 'stdc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library 'supc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
@progress push 5%
Creating makefile for UnrealHeaderTool (no existing makefile)
@progress push 5%
@progress pop
------- Build details --------
Using toolchain located at '/usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '11.0.1' (string), 11 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/usr/local/bin/AirSim/AirSim-main/Unreal/Environments/Blocks/Blocks.uproject" "/usr/local/bin/AirSim/AirSim-main/Unreal/Environments/Blocks/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -abslog="/usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Programs/UnrealBuildTool/Log_UHT.txt"
LogInit: Display: Loading text-based GConfig....
Reflection code generated for BlocksEditor in 2.8289894 seconds
@progress pop
------- Build details --------
Using toolchain located at '/usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '11.0.1' (string), 11 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
ERROR: Building would modify the following engine files:
       
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Binaries/Linux/Android/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Binaries/Linux/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/2D/Paper2D/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/AI/AISupport/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/AI/EnvironmentQueryEditor/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Animation/LiveLink/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Blendables/LightPropagationVolume/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Cameras/CameraShakePreviewer/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Cameras/GameplayCameras/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Compression/OodleData/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Compression/OodleNetwork/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/AnimationSharing/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/CLionSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/CodeLiteSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/GitSourceControl/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/KDevelopSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/NullSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/PerforceSourceControl/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/PlasticSourceControl/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/PluginUtils/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/PropertyAccessNode/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Developer/RiderSourceCodeAccess/Binaries/Linux/UE4Editor.modules
[... 21 lines cut ...]
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/AutomationUtils/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/BackChannel/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/ChaosCloth/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/ChaosClothEditor/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/ChaosEditor/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/ChaosNiagara/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/ChaosSolverPlugin/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/CharacterAI/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/GeometryCache/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/GeometryCollectionPlugin/Binaries/Linux/UE4Editor.modules
[... cut ...]
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Experimental/SkeletalReduction/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/FX/Niagara/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Lumin/MagicLeapMedia/Binaries/Linux/UE4Editor.modules
       /usr/local/bin/Unreal_Engine/UnrealEngine-4.27/Engine/Plugins/Lumin/MagicLeapPassableWorld/Binaries/Linux/UE4Editor.modules

```
Any ideas?

Best,
Gloria

jonyMarino, 2022-08-09:
@Gloriabhsfer have you updated your UE version?

Fergus-MW, 2022-09-14:
Hi Guys, I've got the exact same issue. Was a solution found? I've tried reinstalling Airsim and Unreal to no avail.

tanyaapaul, 2022-10-05:
same here! no updates? Anyone managed to find the error?

einsteinagogo, 2022-10-13:
We have discovered the same issue with UE 4.27 which the documents state should be the version to use has this now changed ?

einsteinagogo, 2022-10-13:
tried 1.8.1 and 1.8.0 both the same in UE 4.27 for Linux

einsteinagogo, 2022-10-13:
and also 1.7.0, 1.6.0 and 1.5.0, on Linux

Gloriabhsfer, 2022-11-01:
> @Gloriabhsfer have you updated your UE version?
Sorry for the late reply. I am now using UE4.27 in Ubuntu 20.04. I figure out this question by reconfig the Uproject for blocks in Linux. Then I can open the block project. The command will be something like:
`ENGINE_FOLDER/Engine/Build/BatchFiles/Linux/Build.sh ExampleProjectEditor Linux DebugGame -project=PROJ_FOLDER/ExampleProject/ExampleProject.uproject -WaitMutex`
More details can be find in this link[https://mercuna.com/compiling-against-multiple-ue4-linux-installed-builds-buildbot/](url). Hope this will solve all the problems.

closed 2022-11-01 by Gloriabhsfer

wuchu1205, 2022-11-17:
> @Gloriabhsfer have you updated your UE version?
> Sorry for the late reply. I am now using UE4.27 in Ubuntu 20.04. I figure out this question by reconfig the Uproject for blocks in Linux. Then I can [...]

What does "ExampleProjectEditor" stands for?

wuchu1205, 2022-11-17:
> @Gloriabhsfer have you updated your UE version?
> Sorry for the late reply. I am now using UE4.27 in Ubuntu 20.04. I figure out this question by reconfig the Uproject for blocks in Linux. Then I can [...]

Hello! I have tried to use this command you mentioned and compile the files successfully. But when I open  Blocks.uproject, it still ask me to rebuild. How to solve this?