microsoft/AirSim #4663: AirSim failed to build with error D8016 on MSVC

fangzhouxia, 2022-08-11:
## Bug report 

- AirSim Version/#commit: 6688d27
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version:  windows server 2019

### What's the issue you encountered?

When we compile AirSim under VS 2019, error D8016 appears. Could you help look?

### Settings

### How can the issue be reproduced?

1.git clone https:https://github.com/microsoft/AirSim F:\gitP\microsoft\AirSim
2. cd F:\gitP\microsoft\AirSim
3. set VSCMD_SKIP_SENDTELEMETRY=1 & "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\Tools\VsDevCmd.bat" -host_arch=amd64 -arch=amd64
4. mkdir build_amd64
5.  cd F:\gitP\microsoft\AirSim\cmake\build_amd64
6. cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.18362.0 -DCMAKE_BUILD_TYPE=Release .. 2>&1
7. msbuild /m /p:Platform=x64 /p:Configuration=Release AirSim.sln /t:Rebuild 2>&1

### Include full error message in text form
14>cl : command line error D8016: '/ZI' and '/GL' command-line options are incompatible [F:\gitP\microsoft\AirSim\cmake\build_amd64\MavLinkCom\MavLinkCom.vcxproj]

Line 66:     15>cl : command line error D8016: '/ZI' and '/GL' command-line options are incompatible [F:\gitP\microsoft\AirSim\cmake\build_amd64\rpclib_wrapper\rpclib\rpc.vcxproj]
	Line 81:          cl : command line error D8016: '/ZI' and '/GL' command-line options are incompatible [F:\gitP\microsoft\AirSim\cmake\build_amd64\MavLinkCom\MavLinkCom.vcxproj]
	Line 89:          cl : command line error D8016: '/ZI' and '/GL' command-line options are incompatible `[F:\gitP\microsoft\AirSim\cmake\build_amd64\rpclib_wrapper\rpclib\rpc.vcxproj]`

Detail log:
[build.log](https://github.com/microsoft/AirSim/files/9307559/build.log)

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2022-08-12:
@fangzhouxia AirSim was upgraded to be compiled with VS 2022

fangzhouxia, 2022-10-10:
ok thanks, I'm using the VS 2022 compiler and it's ok.