microsoft/AirSim #4685: Unable to use SimSetObjectMaterial() and simSetObjectMaterialFromTexture()

Dongzhou-1996, 2022-09-01:
## Bug report 

- AirSim Version/#commit: 1.8.1
- UE/Unity version: 4.27
- autopilot version: N/A
- OS Version: Windows 10

### What's the issue you encountered?

I want to set the material of a StaticMeshActor named `Bennu` by using the API `SimSetObjectMaterial()` and `simSetObjectMaterialFromTexture()`.  The code is provided as follows:

```python
client = airsim.MultirotorClient()
client.confirmConnection()
# get control
client.reset()
client.enableApiControl(True)
# unlock
client.armDisarm(True)

# get target information
target_name = "Bennu"
target = Target(client, target_name=target_name)
print('=> Target name: {} \n=> Target pose: {} \n=> Target size: {}'.format(target.name, target.pose, target.size))
client.simSetObjectMaterialFromTexture(object_name=target_name, texture_path="./sample_texture.jpg")
# client.simSetObjectMaterial(target_name, "Black_Plastic")
```
However, both two API do not work and report following errors (ps: the 'sample_texture.jpg' file is added in my python project directory):
```
Traceback (most recent call last):
  File "E:/Program/airsim_rc_test/remote_control.py", line 198, in <module>
    client.simSetObjectMaterialFromTexture(target_name, "sample_texture.jpg")
  File "D:\Anaconda\envs\airsim\lib\site-packages\airsim\client.py", line 217, in simSetObjectMaterialFromTexture
    return self.client.call("simSetObjectMaterialFromTexture", object_name, texture_path)
  File "D:\Anaconda\envs\airsim\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "D:\Anaconda\envs\airsim\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'simSetObjectMaterialFromTexture' was called with an invalid number of arguments. Expected: 2, got: 3
``` 

### Settings
The settings of StaticMeshActor `Bennu` is shown as follows:
[image]

gabrielpreviato, 2022-09-17:
Same problem here.
Just found out that the problem is that the PyPI version of the AirSim is 1.7.0.
In the 1.8.1 they added a new parameter for both simSetObjectMaterial and simSetObjectMaterialFromTexture.
You can build the python library yourself and then it should work.

gabrielpreviato, 2022-09-19:
@Dongzhou-1996 just update the package, it should work now.