microsoft/AirSim #4708: Wsl2 connection over tcp with airsim

YazdanshensAmin, 2022-09-28:
## Question
### What's your question?
issue with connecting airsim to px4 using wsl2

### Include context on what you are trying to achieve

well I am trying to connect px4 to airsim and for that purpose I followed all the steps in the airsim github page  (https://microsoft.github.io/AirSim/px4_sitl/). PX4 running fine on wsl2 and also airsim works fine on its own but basically I cannot make the TCP connection to work. I was hoping if you could help me to resolve the issue.

#### Context details

### Include details of what you already did to find answers

YazdanshensAmin, 2022-10-05:
That is all the info that I can provide for now

buvnswrn, 2022-10-25:
Did you manage to mak the connection?

droneOTTO, 2023-01-17:
I had the same problem, I solved it by updating the latest release from:  https://github.com/PX4/PX4-Autopilot/releases

 e.g(January 2023):

cd PX4-Autopilot
git checkout v1.13.2
make px4_sitl_default none_iris

2. I also initially bashed PX4 with all the tools and nuttx, not like what is mentioned on the airsim github

git clone https://github.com/PX4/PX4-Autopilot.git --recursive
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh

3. I also opened up all the TCP and UDP ports just in case

YazdanshensAmin, 2023-04-10:
The issue was with my WSL after reinstalling it; it got fixed.

closed 2023-04-10 by YazdanshensAmin