microsoft/AirSim #4712: AirSim Unity crashes when adding Lidar sensor

Dankrupt, 2022-10-05:
## Question
### Whenever I add a lidar sensor in settings.json, the simulator (and Unity) crashes when I try to run it, is this a known issue?

Adding other sensors works without any problems and I can control the drone with QGroundControl through PX4, but the moment I try to add a lidar it breaks.

#### Context details

I'm on Ubuntu 18.04 with Unity version 2021.3.10f1. I know the recommended Unity version is 2019.3.12, however due to an unrelated issue with Unity itself, I cannot open projects in the 2019 version.
I've tried both the latest release from AirSim and v1.6 to see if that would make a difference, but alas. Attached you'll find my settings.json (converted to .txt as i couldn't upload a .json).
[settings.txt](https://github.com/microsoft/AirSim/files/9715831/settings.txt)

abtraore, 2022-10-07:
> ## Question
> ### Whenever I add a lidar sensor in settings.json, the simulator (and Unity) crashes when I try to run it, is this a known issue?
> 
> Adding other sensors works without any problems and I can control the drone with QGroundControl through PX4, but the moment I try to add a lidar it breaks. My lidar settings are quite empty right now, however I have already tried changing stuff there, but this also did not make a difference.
> #### Context details
> 
> I'm on Ubuntu 18.04 with Unity version 2021.3.10f1. I know the recommended Unity version is 2019.3.12, however due to an unrelated issue with Unity itself, I cannot open projects in the 2019 version. I've tried both the latest release from AirSim and v1.6 to see if that would make a difference, but alas. Attached you'll find my settings.json (converted to .txt as i couldn't upload a .json).
> 
> Any help or tips would be greatly appreciated! [settings.txt](https://github.com/microsoft/AirSim/files/9715831/settings.txt)

Lidar sensor in unity doesn't work (yet I guess) you can check the status here : []( https://github.com/microsoft/AirSim/blob/main/docs/unity_api_support.md)

doomgsg, 2022-11-07:
_bigphotodoom airweekly ontario.md FSC®FSC™ 
Duplicate of #

sandilyasg, 2023-11-07:
@abtraore @Dankrupt I removed the "LidarCustom" sensor in the settings.json file for the [single drone with monocular and depth cameras, and lidar](https://microsoft.github.io/AirSim/airsim_tutorial_pkgs/) AirSim ROS Tutorials example. However, when I run airsim_node.launch I get the error "Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id `"front_right_custom_optical" from authority "unknown_publisher" because of a nan value in the transform (0.000000 0.000000 0.000000) (-nan -nan -nan -nan)"`. How did you solve this? I get  the same error for the Two drones, with cameras, lidar, IMU each example after removing the Lidar sensor.