microsoft/AirSim #4725: Issues building airsim_ros_pkgs for ROS 2

bluecamel, 2022-10-28:
## Question
### What's your question?
I've managed to build this in the past, but apparently something has changed.  I'm now getting the following error:

```
-- Found airsim_interfaces: 0.1.0 (/home/bluecamel/DB/AirSim/ros2/install/airsim_interfaces/share/airsim_interfaces/cmake)
CMake Error at /home/bluecamel/DB/AirSim/ros2/install/airsim_interfaces/share/airsim_interfaces/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/bluecamel/DB/AirSim/ros2/install/airsim_interfaces/share/airsim_interfaces/cmake/airsim_interfaces__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /home/bluecamel/DB/AirSim/ros2/install/airsim_interfaces/share/airsim_interfaces/cmake/airsim_interfacesConfig.cmake:41 (include)
  CMakeLists.txt:23 (find_package)
```

This happens on clean installs of ros 2 galactic and rolling (using apt) on Ubuntu 20.04.

### Include context on what you are trying to achieve
N/A

#### Context details
Ubuntu 20.04, Airsim 1.8.1 and main, Unreal 4.27.1

### Include details of what you already did to find answers

alonfaraj, 2022-10-30:
This issue related to cmake version 3.24.
More details [here](https://github.com/ament/ament_cmake/issues/396).
You can downgrade to older versions.

bluecamel, 2022-10-30:
That did it.  Thanks so much!

closed 2022-10-30 by bluecamel