microsoft/AirSim #4734: Build errors for Blocks environment on Ubuntu 20.04

cryptik, 2022-11-04:
## Bug report 

- AirSim Version/#commit: AirSim [main]
- UE/Unity version: 4.27.2
- autopilot version: N/A
- OS Version: Ubuntu 20.04

### What's the issue you encountered?

I cannot load the Blocks environment into the UE4 Editor.  After following the UE4 install steps, I can successfully create a game project using the blueprint feature which I did to validate that UE4 is working

Next, after install of AirSim from the main branch, I can run ./setup.sh and ./build.sh with no issues.  At the end of the script, it reports that the plugin has been built and all environments in Unreal/Environments have been updated. I then launch UE4 and attempt to open the Blocks project.

After selecting the Blocks project, UE4 asks if I want to convert the project and I choose "more options" and "Convert in-place".

The system starts the generate/load process and then dumps messages like the following to the console...

```
[2022.11.03-23.32.28:569][580]Creating makefile for BlocksEditor (no existing makefile)
[2022.11.03-23.32.28:969][580]WARNING: Library 'stdc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
[2022.11.03-23.32.28:969][580]WARNING: Library 'supc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
```
The build then eventually fails resulting in the following....

```
------- Build details --------
Using toolchain located at '/data/tools/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/data/tools/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '11.0.1' (string), 11 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /data/tools/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
ERROR: Building would modify the following engine files:
       
       /data/tools/UnrealEngine/Engine/Binaries/Linux/Android/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Binaries/Linux/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/2D/Paper2D/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/AI/AISupport/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/AI/EnvironmentQueryEditor/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Animation/LiveLink/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Blendables/LightPropagationVolume/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Cameras/CameraShakePreviewer/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Cameras/GameplayCameras/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Compression/OodleData/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Compression/OodleNetwork/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/AnimationSharing/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/CLionSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/CodeLiteSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/GitSourceControl/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/KDevelopSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/NullSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/PerforceSourceControl/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/PlasticSourceControl/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/PluginUtils/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/PropertyAccessNode/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/RiderSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/SubversionSourceControl/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/TextureFormatOodle/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/UObjectPlugin/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Developer/VisualStudioCodeSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Editor/AssetManagerEditor/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Editor/ContentBrowser/ContentBrowserAssetDataSource/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Editor/ContentBrowser/ContentBrowserClassDataSource/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Editor/ContentBrowser/ContentBrowserFileDataSource/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Editor/CryptoKeys/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Editor/CurveEditorTools/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Editor/DataValidation/Binaries/Linux/UE4Editor.modules
       /data/tools/UnrealEngine/Engine/Plugins/Editor/FacialAnimation/Binaries/Linux/UE4Editor.modules
[... 115 lines cut ...]
Generating data for project indexing... 100%
Discovering modules, targets and source code for project...
Generating data for project indexing...   
WARNING: Library 'stdc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library 'supc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
Generating data for project indexing... 100%
Discovering modules, targets and source code for project...
Generating data for project indexing...   
WARNING: Library 'stdc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library 'supc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
Generating data for project indexing... 100%
Writing project files... 100%
```
I can then run `make BlocksEditor` which also generates similar warnings to stdc++ and supc++ as indicated above... but the process does complete and I can open Blocks as an editor project.  I am new to UE and AirSim, so I am wondering why I get the above errors and why I can't compile Blocks into a regular releasable rather than just an editor project.

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

zhjc2014, 2022-11-11:
the same problem

HT-hlf, 2022-11-18:
same problem

sandyUni, 2022-11-24:
same problem

pragatwagle, 2022-11-28:
I am also having this same problem and I am following the same steps.

pragatwagle, 2022-11-28:
Try this: https://github.com/microsoft/AirSim/issues/4535