microsoft/AirSim #4739: Unable to build on M1 Mac

canxkoz, 2022-11-07:
## Bug report 

- AirSim Version/#commit: Latest version as of 6688d27d3712c2a9c824ababec7a2703475b6628
- UE/Unity version: 4.2
- autopilot version: N/A
- OS Version: MacOS 12.6

### What's the issue you encountered?

I can setup.sh and build.sh without any problems but when I open Xcode and build the Blocks.workspace I get the following errors.
"Shell Script Invocation Error Group" and "/Users/Shared/Epic Games/UE_4.27/Engine/Source/Runtime/Engine/Classes/Materials/Material.h:1279:26: Variable 'LayerNames' set but not used"

[image]
### Settings

I don't have any specific settings. But here are some outputs from setup.sh and build.sh

### How can the issue be reproduced?

1. have a mac with an M1 chip, install brew. 
2. go through the setup instructions on [
AirSim MacOS Setup Instructions](https://microsoft.github.io/AirSim/build_macos/)
3. I am stuck on the last step "Open the XCode workspace, and press the Build and run button in the top left." on the setup instructions link. 
### Include full error message in text form
"Shell Script Invocation Error Group" and "/Users/Shared/Epic Games/UE_4.27/Engine/Source/Runtime/Engine/Classes/Materials/Material.h:1279:26: Variable 'LayerNames' set but not used

canxkoz, 2022-11-11:
Deleting the following lines on the screenshot attached on this issue fixed the problem. 1278, 1279, 1280, 1292, 1293, 1294. This specific file is customized as "do not edit" by the developers but deleting these lines fixes the error. I have worked extensively with AirSim so far and I have not experienced any issues due to the deletion.

closed 2022-11-11 by canxkoz

mr-cat-sudo, 2022-11-19:
TYSM!

jdagdelen, 2022-12-04:
I've added a PR that fixes this in a rather simple way. https://github.com/EpicGames/UnrealEngine/pull/9881