microsoft/AirSim #4740: llvm/bin/clang - broken. macOS Ventura

IgorehaV, 2022-11-10:
Hi guys. I've an issue with **./build.sh** and really can't understand where the problem. 
Here is the log from terminal:
`-- Check for working C compiler: /usr/local/opt/llvm/bin/clang
-- Check for working C compiler: /usr/local/opt/llvm/bin/clang - broken
CMake Error at /usr/local/Cellar/cmake/3.24.3/share/cmake/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/usr/local/opt/llvm/bin/clang"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/admin/Documents/AirSim/build_release/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make -f Makefile cmTC_8ccbc/fast && /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_8ccbc.dir/build.make CMakeFiles/cmTC_8ccbc.dir/build
    Building C object CMakeFiles/cmTC_8ccbc.dir/testCCompiler.c.o
    /usr/local/opt/llvm/bin/clang   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.0.sdk -MD -MT CMakeFiles/cmTC_8ccbc.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_8ccbc.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_8ccbc.dir/testCCompiler.c.o -c /Users/admin/Documents/AirSim/build_release/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_8ccbc
    /usr/local/Cellar/cmake/3.24.3/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8ccbc.dir/link.txt --verbose=1
    /usr/local/opt/llvm/bin/clang  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.0.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/cmTC_8ccbc.dir/testCCompiler.c.o -o cmTC_8ccbc 
    ld: library not found for -lSystem
    clang-15: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_8ccbc] Error 1
    make: *** [cmTC_8ccbc/fast] Error 2
    
  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!`

My Mac is on Ventura OS. The latest Xcode version and seems to be the latest Xcode Command tools base on **xcode-select install** command.
-- The C compiler identification is Clang 15.0.3
-- The CXX compiler identification is Clang 15.0.3

I tried to changes **llvm** to **llvm@15** in the **setup.sh** but it doesn't help me. The same error still appears. 

Best regards

deonblaauw, 2022-11-23:
I managed to fix this today by using `which clang` and `which clang++` to get the path where the compilers are installed, then I changed my build.sh file (under AirSim/Unity). Mine were both under `/usr/bin/` , this is the section in the build.sh file I changed:

```cd linux-build
if [ "$(uname)" == "Darwin" ]; then
    export CC=/usr/bin/clang
    export CXX=/usr/bin/clang++
else
    export CC="clang-8"
    export CXX="clang++-8"
fi
```

imccs9911, 2023-07-06:
Same problem 😒

Mio-Atse, 2024-10-20:
> I had a similar problem when trying to build for Unity using the build.sh file under AirSim/Unity.
> 
> I managed to fix this today by using `which clang` and `which clang++` to get the path where the compilers are installed, then I changed my build.sh file (under AirSim/Unity). Mine were both under `/usr/bin/` , this is the section in the build.sh file I changed:
> 
> ```
> if [ "$(uname)" == "Darwin" ]; then
>     export CC=/usr/bin/clang
>     export CXX=/usr/bin/clang++
> else
>     export CC="clang-8"
>     export CXX="clang++-8"
> fi
> ```
> 
> I also had to do `sudo xcode-select --install` to get the script to work (remember to restart your terminal after using this command)

Thanks it works!

Also you may need to clear cmake and cache files with following, then ./setup.sh and ./build.sh again.

rm -f CMakeCache.txt
rm -rf CMakeFiles/