microsoft/AirSim #4754: msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'simGetImages' was called with an invalid number of arguments. Expected: 2, got: 3

zclyaya, 2022-11-29:
C:\Users\zcl\.conda\envs\zclpytorch\python.exe D:\Fixedwing-Airsim-master\src\main.py 

In file d:/JSBSim/aircraft/x8/x8.xml: line 80
Engine location ignored, only thruster location is used.

In file d:/JSBSim/aircraft/x8/x8.xml: line 85
Engine orientation ignored, only thruster orientation is used.
Traceback (most recent call last):
  File "D:\Fixedwing-Airsim-master\src\main.py", line 217, in <module>
    run_simulator()
  File "D:\Fixedwing-Airsim-master\src\main.py", line 195, in run_simulator
    env.simulation_loop(rectangle)
  File "D:\Fixedwing-Airsim-master\src\main.py", line 92, in simulation_loop
    image.get_np_image(image_type=airsim.ImageType.Scene)
  File "D:\Fixedwing-Airsim-master\src\image_processing.py", line 44, in get_np_image
    False)])
  File "C:\Users\zcl\.conda\envs\zclpytorch\lib\site-packages\airsim\client.py", line 264, in simGetImages
    responses_raw = self.client.call('simGetImages', requests, vehicle_name)
  File "C:\Users\zcl\.conda\envs\zclpytorch\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\zcl\.conda\envs\zclpytorch\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'simGetImages' was called with an invalid number of arguments. Expected: 2, got: 3

     JSBSim Flight Dynamics Model v1.1.11 [GitHub build 741/commit 10e077b486c42d38e76d1dd7f0aeb744f92d96a0] Feb 13 2022 12:30:25
            [JSBSim-ML v2.0]

JSBSim startup beginning ...

Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Process finished with exit code 1

airsim==1.5.0    python 4.5.1

I do not how to solve this problem, can you help me ?

peach-water, 2023-01-01:
I have the same problem.

MaltheRaschke, 2023-03-21:
I have the same problem when running the Event camera "test_event_sim.py", but not when running other scripts as hello_drone.py - both work in blocks enviroment, but not in a custom enviroment I've downloaded.
Any inputs?

zclyaya, 2023-03-21:
> 我有同样的问题。
你好，请问解决这个问题了吗？

MaltheRaschke, 2023-03-21:
I fixed it with replacing the Plugins folder from a working environment (the "Blocks" environment) with the non-working environment's Plugins folder