microsoft/AirSim #4755: Whether the weather set by AirSim has physical effect on lidar?

asaltedfish4, 2022-11-30:
## Question
### What's your question?

Whether the weather set by AirSim has physical effect on lidar?

### Include context on what you are trying to achieve

I set up the weather via AirSim. And I want to know whether the weather (rain or snow) affect the point cloud data collected by lidar and the principle of lidar collecting point cloud data.
Thank you very much!

dirkmcpherson, 2023-01-30:
Did you find an answer for this? It doesn't seem like the weather effects the sensors (Barometer.pressure anyway)

asaltedfish4, 2023-03-21:
> Did you find an answer for this? It doesn't seem like the weather effects the sensors (Barometer.pressure anyway)

I found that the weather doesn't effect the lidar because in UE only the object who has collision can be detect by lidar.