microsoft/AirSim #4756: Airsim error: Not receiving any messages from HIL, please restart your HIL node and try again

AshishSingh-Glitch90, 2022-11-30:
## Question
### What's your question?

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

Hi, 
I am trying to implement flight simulation using Airsim hardware in the loop(HITL), px4 version 1.9.2 on pixhawk 2.4.8 by programming pixhawk using Qgroundcontrol which I was able to work.
Now I want to implement the same with the Xbox Controller for which I followed:

[https://microsoft.github.io/AirSim/xbox_controller/](url)
and 
[https://docs.px4.io/main/en/simulation/hitl.html](url)

using this "MavLinkTest.exe -serial:*,115200 -proxy:127.0.0.1:14550 -server:127.0.0.1:14570" command I am able to build a 
connect between the QGroundControl and Proxy but Airsim shows a message to restart the HITL node as shown below: 
  
[image]

[image]

My Airsim Json settings are:

{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "ViewMode": "",
    "ClockType": "SteppableClock",
    "Vehicles": {
      "PX4": {
          "VehicleType": "PX4Multirotor",
          "SitlIp": "",
          "SitlPort": 14560,
          "UdpIp": "127.0.0.1",
          "UdpPort": 14570,
          "UseSerial": false,
          "LockStep": true,
        "Parameters": {
            "NAV_RCL_ACT": 0,
            "NAV_DLL_ACT": 0,
            "COM_OBL_ACT": 1,
            "LPE_LAT": 47.641468,
            "LPE_LON": -122.140165
        }
    }
}
}

Do you have any suggestion how I can fix this.

Thanks

enesvta, 2025-02-08:
Did you solve this question? I tried everything but I can't figure it out

AshishSingh-Glitch90, 2025-02-08:
@enesvta no I couldn't solve it.

https://www.youtube.com/@dustin5363

This guy shows that he is able to do it, and I tried reaching out to him but no response.