microsoft/AirSim #4761: How to fix this error? C1900

FatalJack, 2022-12-09:
[image]

I cannot do anything because of this question.

FatalJack, 2022-12-09:
It's not because of the version of UE. I tried 4.27 before and have met same error, three lines and boom.

nagsterFVZ, 2022-12-14:
Was having this same issue with UE 4.27 and VS2022.
To solve it I opened the project in unreal first and and changed the VS version.
You can do this under Edit > Editor Preferences > Source Code Editor
Set that to Visual Studio 2022, then under file generate the visual studio project.

Now I was able to add the AirSim plugin to my project and build it as usual.

huangzhuofei, 2023-01-01:
> Was having this same issue with UE 4.27 and VS2022. To solve it I opened the project in unreal first and and changed the VS version. You can do this under Edit > Editor Preferences > Source Code [...]

Thank you for helping me a lot

RyanJayWu, 2023-01-08:
Same problem... Anyone can help solve this ?

JackeyLea, 2023-11-14:
> Was having this same issue with UE 4.27 and VS2022. To solve it I opened the project in unreal first and and changed the VS version. You can do this under Edit > Editor Preferences > Source Code [...]

Thx