microsoft/AirSim #4765: AirSim Blocks Build Error C1900: Il mismatch between 'P1' version '20220715' and 'P2' version '20210202'

dr-chris-fink, 2022-12-19:
## Question
### What's your question?
AirSim builds but the Blocks project does not. I get errors regarding a P1 vs. P2 mismatch, which (from web searches) seems to indicate that AirSim should not have been built with the /GL flag.  Not sure which build script controls this.

### Include context on what you are trying to achieve

Tried to follow instructions here: https://microsoft.github.io/AirSim/build_windows/

#### Context details

See attachments. I did try to re-target the Blocks solution to use 19041, with the same result.

### Include details of what you already did to find answers
[AirSimBuild.log](https://github.com/microsoft/AirSim/files/10262013/AirSimBuild.log)
[BlocksBuild.log](https://github.com/microsoft/AirSim/files/10262014/BlocksBuild.log)
[BlocksRebuild19041.log](https://github.com/microsoft/AirSim/files/10262015/BlocksRebuild19041.log)

slowdancer1, 2022-12-26:
same problem... have you solved them?

Lychannnn, 2023-03-17:
have you solved them?

dr-chris-fink, 2023-03-17:
Yes –

Here’s the response I got:

“Thanks for all the details. I think I understand what's going on. You are
building airlib with vs2022 and its associations and blocks with vs2019 and
its. Here is a description of how to move the project blocks from vs2019 to
vs2022: https://microsoft.github.io/AirSim/build_windows/#how-to-use-airsim
<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fmicrosoft.github.io%2FAirSim%2Fbuild_windows%2F%23how-to-use-airsim&data=05%7C01%7Cv-jomarino%40microsoft.com%7C727010c60b7a48ecb08e08daf00f9b29%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638086247082025652%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=it2Pf3BW78srmb1FQFIP8qlJAtVc5pZjEzgDPstfeOo%3D&reserved=0>
. Look at the note there.”

I replied:

“Thanks, but actually my intent is to stay with 2019 if possible, as I have
other projects and plugins which require a version 2019 build. So rather
than move Blocks to vs2022,  is there a way to reconfigure the AirSim build
scripts (or the VS2022 Developer Command Prompt) to use VS2019?”

Their response:

“Anything before this commit works with VS2019:
https://github.com/microsoft/AirSim/pull/4486/commits/b3913c8ff9405204f7be8aff4e1c3b67f3362f0a

So you can for example checkout ·  v1.7.0-windows
<https://github.com/microsoft/AirSim/releases/tag/v1.7.0-windows>:   git
checkout v1.7.0-windows “

--------------------------------

Dr. Christopher Fink, Ph.D.

Senior Physicist

JRM Technologies, Inc.

***@***.***

540-736-8106

*From:* Lychannnn ***@***.***>
*Sent:* Friday, March 17, 2023 5:55 AM
*To:* microsoft/AirSim ***@***.***>
*Cc:* dr-chris-fink ***@***.***>; Author <
***@***.***>
*Subject:* Re: [microsoft/AirSim] AirSim Blocks Build Error C1900: Il
mismatch between 'P1' version '20220715' and 'P2' version '20210202' (Issue
#4765)

have you solved them?