microsoft/AirSim #4780: Multirotor Y-axis drift but not X-axis

WreckItTim, 2023-01-25:
## Bug report 

- AirSim Version/#commit: 1.8.1
- UE/Unity version: N/A - using release blocks v1.8.1 Windows
- autopilot version: N/A
- OS Version: Windows 10 Pro

### What's the issue you encountered?

Noticed a drift along the Y-axis in multirotor movement when calling moveByVelocityAsync().join()
However not apparent along the X-axis

### Settings

{
  'ClockSpeed': 8,
  "SimMode": "Multirotor",
  "ViewMode": "SpringArmChase",
  "SubWindows": []
}

### How can the issue be reproduced?

I made this script to just plug and play and repeat error. Run this python script (replacing as needed to launch configuration). I used the v1.8.1 Blocks release on Windows and dropped the parent folder in the same folder as the bellow python script file, along with setup_path.py. After launching, you can use the terminal to input commands. 

1. Enter 'y' to line up with the y-axis then 'f' to move forward. Watch. After moving, the drone will begin to spin in a slow circle. Press 'f' to go forward again. The circle will make the movement forward look like a drift. 
2. Now enter 'x' to line up with the x-axis then 'f' to move forward. Watch. The drone will not spin in a circle. Press 'f' to go forward again, no apparent drift. 
3. Line up with the y-axis again and go forward. The same spin and drift will reappear!
4. Repeat as much as you want to make sure you're not crazy. 

Note that reducing the clock speed will have the same bug but it will be slower and harder to see, but trust me it is still there if you look close enough.

```
import subprocess
import setup_path
import airsim
import json
import os

release_path = 'Blocks/WindowsNoEditor/Blocks.exe'

flags = '-Windowed'

settings = {
  'ClockSpeed': 8,
  "SimMode": "Multirotor",
  "ViewMode": "SpringArmChase",
  "SubWindows": []
}
settings_path = os.getcwd() + '/settings.json'
json.dump(settings, open(settings_path, 'w'), indent=2)

terminal_command = f'{release_path} {flags} -settings=\"{settings_path}\"'
process = subprocess.Popen(terminal_command, creationflags=subprocess.CREATE_NEW_PROCESS_GROUP)

client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)
client.takeoffAsync().join()
client.moveByVelocityAsync(0, 0, -1, 2).join()

facing = ''

def rotate_to(yaw):
	client.rotateToYawAsync(yaw) 
		
def move(x_speed, y_speed, z_speed, duration):
	client.moveByVelocityAsync(x_speed, y_speed, z_speed, duration).join()

while True:
	print('x to rotate facing x-axis')
	print('y to rotate facing y-axis')
	print('f to move forward')
	response = input()
	if response not in ['x', 'y', 'f']:
		print('invalid input')
		continue
	if response == 'x':
		rotate_to(180)
		facing = 'x'
	if response == 'y':
		rotate_to(270)
		facing = 'y'
	if response == 'f':
		if facing == 'x':
			move(-2, 0, 0, 2)
		elif facing == 'y':
			move(0, -2, 0, 2)
		else:
			print('must enter x or y first')
```

### Include full error message in text form
no error message

**What's better than filing an issue? Filing a pull request :).**

WreckItTim, 2023-03-16:
Found that a better stop-gap is to either sleep until oscillations stop (expensive) or force the velocity of the drone to 0 in all dims, by adding something like this to where it is needed in your code:

```
# stabalize drone
if stabelize:
	self._airsim._client.rotateByYawRateAsync(0, 0.1)
	self._airsim._client.moveByVelocityAsync(0, 0, 0, 0.1).join()
```

sandilyasg, 2024-02-03:
@WreckItTim Can you please share how you set the different PID gains? Is it using the API calls or directly in AirSim/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/Params.hpp?