microsoft/AirSim #4792: LiDAR Camera mis-projection

pravinjaisawal, 2023-02-15:
## Question
### What's your question?

For my work, I record multiple camera images and LiDAR data at 10 frames per second. For this purpose, I move the drone at a speed of 5 m/s and have set the clock rate to 0.05.  The whole recording process works without any problems. After recording, I use the camera position and LiDAR position from the settings to calculate the transformation matrix. Based on the settings, the transformation matrix is:

T = [0.0 1.0 0.0 0.0
      0.0 0.0 1.0 -0.4 
      1.0 0.0 0.0 -0.2 
      0.0 0.0 0.0 1.0]
When I use this transformation matrix to project LiDAR point clouds onto camera images, sometimes it works fine, but sometimes it results in misprojections. 
- Example: correctly projected
[image]

- Example Incorrectly projected
[image]

Questions:
 - Does the relative camera - LiDAR poses change during recording? If yes, how to fixed their relative poses?
 - Do you have other suggestions to resolve this issue?
### Include context on what you are trying to achieve

#### Context details

- AirSim Version/#commit:1.7
- UE/Unity version:4.27
- autopilot version:
- OS Version: Ubuntu 20
- [airsim_settings](https://gist.github.com/pravinjaisawal/849d244c955b3d0dbebf06b9bc2b51af)

closed 2023-05-03 by pravinjaisawal

sandilyasg, 2023-11-02:
@pravinjaisawal I see the issue is closed, can you please share how you fixed it?

pravinjaisawal, 2023-11-03:
@sandilyasg I recorded the camera and LiDAR pose for every frame, then calculate the transformation matrix. This matrix maps the LiDAR point cloud properly into camera image