microsoft/AirSim #4794: could not build AirSim on windows

mgrallos, 2023-02-16:
## Question
### What's your question?
Hi I am trying to build AirSim on Windows 10. And while trying to run build.cmd these errors occur 

CMake Error: Could not create named generator Visual Studio 17 2022

Generators
  Visual Studio 15 2017 [arch] = Generates Visual Studio 2017 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 14 2015 [arch] = Generates Visual Studio 2015 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 12 2013 [arch] = Generates Visual Studio 2013 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 11 2012 [arch] = Generates Visual Studio 2012 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 10 2010 [arch] = Generates Visual Studio 2010 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Visual Studio 9 2008 [arch]  = Generates Visual Studio 2008 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Visual Studio 8 2005 [arch]  = Deprecated.  Generates Visual Studio 2005
                                 project files.  Optional [arch] can be
                                 "Win64".
  Borland Makefiles            = Generates Borland makefiles.
  NMake Makefiles              = Generates NMake makefiles.
  NMake Makefiles JOM          = Generates JOM makefiles.
  Green Hills MULTI            = Generates Green Hills MULTI files
                                 (experimental, work-in-progress).
  MSYS Makefiles               = Generates MSYS makefiles.
  MinGW Makefiles              = Generates a make file for use with
                                 mingw32-make.
  Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  Watcom WMake                 = Generates Watcom WMake makefiles.
  CodeBlocks - MinGW Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles JOM
                               = Generates CodeBlocks project files.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - MinGW Makefiles   = Generates CodeLite project files.
  CodeLite - NMake Makefiles   = Generates CodeLite project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Sublime Text 2 - MinGW Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - NMake Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                               = Generates Sublime Text 2 project files.
  Kate - MinGW Makefiles       = Generates Kate project files.
  Kate - NMake Makefiles       = Generates Kate project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Eclipse CDT4 - NMake Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - MinGW Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.

Error: could not load cache
Error: could not load cache

#### Build failed - see messages above.

Can you let me know what these errors are

xxEoD2242, 2023-02-16:
Hey @mgrallos,

You need to install VS 2022 to get that to work. Also, this repo will be shutdown soon by Microsoft.

We continued the open-source version here: [Colosseum](https://github.com/CodexLabsLLC/Colosseum) with support UE5 and UE4.27 as well.

mgrallos, 2023-02-16:
> Hey @mgrallos,
> 
> You need to install VS 2022 to get that to work. Also, this repo will be shutdown soon by Microsoft.
> 
> We continued the open-source version here: [Colosseum](https://github.com/CodexLabsLLC/Colosseum) with support UE5 and UE4.27 as well.

hi I already did install VS 17 2022. But what work for me is upgrading the cmake version to latest. Also, thanks for the link of the new repository. Appreciate it!

christian33moreno, 2023-02-17:
@xxEoD2242 Hello! Would I be able to run deep reinforcement learning python script to train the drone agent using Colosseum?

xxEoD2242, 2023-02-17:
@christian33moreno hey!

Yes, you can! There is an integration with the gym library to run Deep RL algorithms. We also have a slack channel if you run into any issues. Let me know if you need any help getting started!

christian33moreno, 2023-02-17:
@xxEoD2242 I'll need help integrating the python script with the simulation. Is there documentation or some kind of video that can walk me through how to do it? Thanks so much, I really appreciate the response.

mgrallos, 2023-02-17:
That might be useful for me too in the near future.

xxEoD2242, 2023-02-17:
@christian33moreno @mgrallos Sounds good! Happy to help. Feel free to reach out on GitHub or email me at tylerfedrizzi@codex-labs-llc.com

My company is also building a new tool to help with this called the [SWARM Developer Platform](https://www.swarmsim.io/overview/developer) that has automated RL integration so you can run in the cloud, iterate and read back data quickly, then transfer to a real world system in a few clicks through our deployment process with PX4, ROS and more. If you're interested, drop me a line in the email

xxEoD2242, 2023-02-17:
> @xxEoD2242 I'll need help integrating the python script with the simulation. Is there documentation or some kind of video that can walk me through how to do it? Thanks so much, I really appreciate the response.

As for documentation, we don't have anything yet but I will put this on the list!

cq68435793, 2023-03-28:
Have you got a meal yet

cameroncrose72, 2023-04-02:
@mgrallos I had the exact same issue, and while the build is not yet complete as I type this (I'm just so excited), it seems to be working!