microsoft/AirSim #4796: Problem to run <build.cmd> about CMake for Microsoft Airsim

agam002, 2023-02-19:
When I followed the process of build AirSim, There were some problems.
put the text build.cmd on command line, I got the message like this.
I'm not familiar in Visual Studio & Programming like this please help me.
=========================================
 
C:\Users\arion\source\repos\AirSim>build.cmd
found Powershell
Found cmake version: 3.24.202208181-MSVC_2
Starting cmake to build rpclib...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "C:/Users/arion/source/repos/AirSim/external/rpclib/rpclib-2.3.0/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/arion/source/repos/AirSim/external/rpclib/rpclib-2.3.0/build/CMakeFiles/CMakeError.log".
MSBuild version 17.4.1+9a89d02ff for .NET Framework
MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj
MSBuild version 17.4.1+9a89d02ff for .NET Framework
MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj

#### Build failed - see messages above.

The following link is the AirSim git repository
https://github.com/Microsoft/AirSim

Many thanks for helping!

dianzizs, 2023-11-22:
I have the same issue with you. Did you figure out how to address this problem? Thanks a lot

dianzizs, 2023-11-22:
> # When I followed the process of build AirSim, There were some problems.
> put the text build.cmd on command line, I got the message like this.
> I'm not familiar in Visual Studio & Programming like [...]
I have the same issue with you. Did you figure out how to address this problem? Thanks a lot

codex-exe, 2024-02-21:
I was facing the same issue but in my case i had set "cl" environment variable to something for another project. Removing that variable made it work.