microsoft/AirSim #4805: compilation failed

denk1, 2023-03-03:
Does anybody know why this happens while compilation?
Platform: Ubuntu 22.04

[4/6] Link (lld) libUE4Editor-AirSim.so
ld.lld: error: undefined symbol: msr::airlib::CarRpcLibServer::CarRpcLibServer(msr::airlib::ApiProvider*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, unsigned short)
>>> referenced by SimModeCar.cpp:86 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/Vehicles/Car/SimModeCar.cpp:86)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(ASimModeCar::createApiServer() const)

ld.lld: error: undefined symbol: msr::airlib::RpcLibServerBase::RpcLibServerBase(msr::airlib::ApiProvider*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned short)
>>> referenced by SimModeComputerVision.cpp:24 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/Vehicles/ComputerVision/SimModeComputerVision.cpp:24)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(ASimModeComputerVision::createApiServer() const)

ld.lld: error: undefined symbol: msr::airlib::MultirotorRpcLibServer::MultirotorRpcLibServer(msr::airlib::ApiProvider*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, unsigned short)
>>> referenced by SimModeWorldMultiRotor.cpp:85 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/Vehicles/Multirotor/SimModeWorldMultiRotor.cpp:85)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(ASimModeWorldMultiRotor::createApiServer() const)

ld.lld: error: undefined symbol: common_utils::FileSystem::createDirectory(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
>>> referenced by FileSystem.hpp:63 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/common_utils/FileSystem.hpp:63)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(common_utils::FileSystem::getLogFolderPath(bool, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&))
>>> referenced by FileSystem.hpp:69 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/common_utils/FileSystem.hpp:69)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(common_utils::FileSystem::ensureFolder(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&))
>>> referenced by FileSystem.hpp:63 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/common_utils/FileSystem.hpp:63)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(common_utils::FileSystem::getLogFileNamePath(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool))
>>> referenced 1 more times

ld.lld: error: undefined symbol: common_utils::FileSystem::getExecutableFolder()
>>> referenced by Settings.hpp:57 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/Settings.hpp:57)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(msr::airlib::Settings::getExecutableFullPath(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld.lld: error: undefined symbol: common_utils::FileSystem::getUserDocumentsFolder()
>>> referenced by FileSystem.hpp:57 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/common_utils/FileSystem.hpp:57)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(common_utils::FileSystem::getAppDataFolder())

ld.lld: error: undefined symbol: mavlinkcom::UdpSocket::bind(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int)
>>> referenced by ArduRoverApi.hpp:148 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/vehicles/car/firmwares/ardurover/ArduRoverApi.hpp:148)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(msr::airlib::ArduRoverApi::connect())
>>> referenced by ArduCopterApi.hpp:324 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/arducopter/ArduCopterApi.hpp:324)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(msr::airlib::ArduCopterApi::connect())

ld.lld: error: undefined symbol: mavlinkcom::UdpSocket::sendto(void const*, unsigned long, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned short)
>>> referenced by ArduRoverApi.hpp:302 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/vehicles/car/firmwares/ardurover/ArduRoverApi.hpp:302)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(msr::airlib::ArduRoverApi::sendSensors())
>>> referenced by ArduCopterApi.hpp:476 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/arducopter/ArduCopterApi.hpp:476)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(msr::airlib::ArduCopterApi::sendSensors())

ld.lld: error: undefined symbol: msr::airlib::MultirotorApiBase::waitForFunction(std::__1::function<bool ()>, float)
>>> referenced by MavLinkMultirotorApi.hpp:482 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp:482)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(msr::airlib::MavLinkMultirotorApi::land(float))
>>> referenced by MavLinkMultirotorApi.hpp:410 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp:410)
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(msr::airlib::MavLinkMultirotorApi::waitForHomeLocation(float))
>>> referenced by MavLinkMultirotorApi.hpp:425 (/home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp:425)
[... cut ...]
>>>               /home/den/Projects/AirSim-1.8.0-linux/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(msr::airlib::MavLinkMultirotorApi::createMavSerialConnection(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int))

ld.lld: error: undefined symbol: mavlinkcom::MavLinkNode::connect(std::__1::shared_ptr<mavlinkcom::MavLinkConnection>)
LogInit: Warning: Still incompatible or missing module: Blocks
LogInit: Warning: Still incompatible or missing module: AirSim

closed 2023-03-04 by denk1

kalind-seneos, 2023-05-18:
What was the solution, I'm encountering a similar issue ?

denk1, 2023-05-18:
I moved  from  gcc to clang

ArciAndres, 2024-09-17:
Excuse me, how does this relate specifically with the suggested ./setup.sh and ./build.sh scripts?