microsoft/AirSim #4811: Can't find copter camera sensor size

Chepa14, 2023-03-17:
## Question
### What's your question?
Where I can find the sensor size (width and height) of the drone virtual camera?

### Include context on what you are trying to achieve
I'm trying to analyze images that I get from the bottom center camera (number 3), but I need to know the camera sensor size.

#### Context details
I'm using [v1.8.1 - Linux](https://github.com/microsoft/AirSim/releases/tag/v1.8.1) AirSimNH as a virtual environment with Ardupilot ArduCopter.
Here's my setting.json:
{
   "SettingsVersion":1.2,
   "LogMessagesVisible":true,
   "SimMode":"Multirotor",
   "Vehicles":{
      "Copter":{
         "VehicleType":"ArduCopter",
         "UseSerial":false,
         "DefaultVehicleState":"Armed",
         "LocalHostIp":"127.0.0.1",
         "UdpIp":"127.0.0.1",
         "UdpPort":9003,
         "ControlPort":9002,
         "AllowAPIAlways": true,
      }
   },
   "CameraDefaults":{
      "CaptureSettings":[
         {
            "ImageType":0,
            "Width":600,
            "Height":600,
            "FOV_Degrees":90,
            "AutoExposureSpeed":100,
            "AutoExposureBias":0,
            "AutoExposureMaxBrightness":0.64,
            "AutoExposureMinBrightness":0.03,
            "MotionBlurAmount":0,
            "TargetGamma":1.0,
            "ProjectionMode":"",
            "OrthoWidth":10
         }
      ],
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
   }
}
### Include details of what you already did to find answers
I searched through airsim, unreal engine, and ardupilot projects for any information, but found only setup for cinematicCamera. But I'm not sure that it's the correct camera used on copter...

xxEoD2242, 2023-03-17:
@Chepa14 I believe that the camera is 640 x 480 (w x h), as that is the default settings. I can point you to the source code if you like.

Also, Microsoft archived this project so we continued it and renamed Colosseum. You can check it out here: [Colosseum Robotics Simulator](https://github.com/CodexLabsLLC/Colosseum)

Chepa14, 2023-03-17:
@xxEoD2242 Yes, please.  `640 x 480 (w x h)` - is it size of sensor (inner matrix of a camera)?

xxEoD2242, 2023-03-17:
Oh, I was wrong 240 x 144.

[AirSimSettings.hpp#L135](https://github.com/CodexLabsLLC/Colosseum/blob/052228ae6c9a03ba5ccd2ffd6422f1433588302b/AirLib/include/common/AirSimSettings.hpp#L135)

Chepa14, 2023-03-17:
@xxEoD2242 This is a configuration for image capturing, so it's image size, but not a sensor. Look at this article to understand what I mean https://capturetheatlas.com/camera-sensor-size/

xxEoD2242, 2023-03-17:
Ah, gotcha. 

Sensor size is 23.76 mm (width) and 13.365 mm (height). You have to open BP_PIPCamera.uasset in the UE Editor and then look at Filmback settings.

Chepa14, 2023-03-20:
@xxEoD2242 Thanks for the answer! Can you please paste a full screen of the camera parameters and how they look in UE Editor?

closed 2023-03-23 by Chepa14