microsoft/AirSim #4812: Where should I put python file for Airsim simulation?

Charles-Lim93, 2023-03-20:
## Question
### What's your question?

How I can connect Python client to Airsim in Window? I'm a new guy in here and very confused to where to my python client. 
About client, and all that kinds of stuff. It might be very easy question but cannot not find clear example in google...
Is there any supplements that I can use for Airsim with python?

Thanks in advance.

[image]

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

Aklice-new, 2023-04-01:
You can put you python file anywhere ,and the airsim connect the by net 
[image]
，but make sure that you have the python enviroment for airsim(pip install airsim). It worked on my computer , I'm also a new guy.
Hope that will help you!

Anon777098, 2024-02-27:
Hey guys, I have managed to connect python client with AirSim. But when I run it, each time it asks 'Press any key for camera parameters' even when I change the code to do something different. Do you have any solution for this? Thanks