microsoft/AirSim #4818: run blocks error on ubuntu 20.04

blackcyavuz, 2023-03-29:
## Bug report 

- AirSim Version/#commit:
- UE/Unity version:4.27
- autopilot version:
- OS Version:ubuntu20.04

The project could not be compiled. Would you like to open it in Null Source Code Access?

Running /home/yavuzselim/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project="/home/yavuzselim/AirSim/Unreal/Environments/Blocks (copy)/Blocks.uproject" -TargetType=Editor -Progress -NoEngineChanges -NoHotReloadFromIDE

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine/Engine ~/UnrealEngine/Engine/Binaries/Linux
Using 'git status' to determine working set for adaptive non-unity build (/home/yavuzselim/UnrealEngine).
Creating makefile for BlocksEditor (no existing makefile)
WARNING: Library 'stdc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library 'supc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
@progress push 5%
Creating makefile for UnrealHeaderTool (no existing makefile)
@progress push 5%
@progress pop
------- Build details --------
Using toolchain located at '/home/yavuzselim/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/yavuzselim/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '11.0.1' (string), 11 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /home/yavuzselim/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/home/yavuzselim/AirSim/Unreal/Environments/Blocks (copy)/Blocks.uproject" "/home/yavuzselim/AirSim/Unreal/Environments/Blocks (copy)/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -abslog="/home/yavuzselim/UnrealEngine/Engine/Programs/UnrealBuildTool/Log_UHT.txt"
LogInit: Display: Loading text-based GConfig....
Reflection code generated for BlocksEditor in 2.5044659 seconds
@progress pop
------- Build details --------
Using toolchain located at '/home/yavuzselim/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/yavuzselim/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '11.0.1' (string), 11 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /home/yavuzselim/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
ERROR: Building would modify the following engine files:
       
       /home/yavuzselim/UnrealEngine/Engine/Binaries/Linux/Android/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Binaries/Linux/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/2D/Paper2D/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/AI/AISupport/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/AI/EnvironmentQueryEditor/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Animation/LiveLink/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Blendables/LightPropagationVolume/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Cameras/CameraShakePreviewer/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Cameras/GameplayCameras/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Compression/OodleData/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Compression/OodleNetwork/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/AnimationSharing/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/CLionSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/CodeLiteSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/GitSourceControl/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/KDevelopSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/NullSourceCodeAccess/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/PerforceSourceControl/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/PlasticSourceControl/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/PluginUtils/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/PropertyAccessNode/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Developer/RiderSourceCodeAccess/Binaries/Linux/UE4Editor.modules
[... 34 lines cut ...]
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Experimental/PlatformCrypto/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Experimental/PythonScriptPlugin/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Experimental/SkeletalReduction/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/FX/Niagara/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Lumin/MagicLeapMedia/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Lumin/MagicLeapPassableWorld/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Media/AndroidMedia/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Media/AvfMedia/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Media/ImgMedia/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Media/MediaCompositing/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Media/MediaPlayerEditor/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Media/WebMMedia/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Media/WmfMedia/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/MeshPainting/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Messaging/TcpMessaging/Binaries/Linux/UE4Editor.modules
       /home/yavuzselim/UnrealEngine/Engine/Plugins/Messaging/UdpMessaging/Binaries/Linux/UE4Editor.modules

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

DHrobotman, 2023-04-08:
I have the same problem when I ran the demo in /AirSim/Umreal/Environments/Blocks  on ubuntu18.04 , have you sovle the problem ?

wagh311, 2023-07-13:
I have the same problem，have you sovle the problem ?

prncoprs, 2025-05-13:
Same problem.