microsoft/AirSim #4822: Build command cmake error AirSim

agam002, 2023-04-03:
## Question
### What's your question?
Hi , 
I got this error while trying to run the build command on VS command prompt as admintrator. This is how the error looked like : 

**************************************************                                                           ********************
** Visual Studio 2022 Developer Command Prompt v17                                                           .5.3
** Copyright (c) 2022 Microsoft Corporation
**************************************************                                                           ********************

C:\Windows\System32>cd AirSim

C:\Windows\System32\AirSim>build.cmd
found Powershell
Found cmake version: 3.14.7
Starting cmake to build rpclib...
CMake Error: Could not create named generator Visual Studio 17 2022

Generators
* Visual Studio 16 2019        = Generates Visual Studio 2019 project files.
                                 Use -A option to specify architecture.
  Visual Studio 15 2017 [arch] = Generates Visual Studio 2017 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 14 2015 [arch] = Generates Visual Studio 2015 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 12 2013 [arch] = Generates Visual Studio 2013 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 11 2012 [arch] = Generates Visual Studio 2012 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 10 2010 [arch] = Generates Visual Studio 2010 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Visual Studio 9 2008 [arch]  = Generates Visual Studio 2008 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Borland Makefiles            = Generates Borland makefiles.
  NMake Makefiles              = Generates NMake makefiles.
  NMake Makefiles JOM          = Generates JOM makefiles.
  Green Hills MULTI            = Generates Green Hills MULTI files
                                 (experimental, work-in-progress).
  MSYS Makefiles               = Generates MSYS makefiles.
  MinGW Makefiles              = Generates a make file for use with
                                 mingw32-make.
  Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  Watcom WMake                 = Generates Watcom WMake makefiles.
  CodeBlocks - MinGW Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles JOM
                               = Generates CodeBlocks project files.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - MinGW Makefiles   = Generates CodeLite project files.
  CodeLite - NMake Makefiles   = Generates CodeLite project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Sublime Text 2 - MinGW Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - NMake Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                               = Generates Sublime Text 2 project files.
  Kate - MinGW Makefiles       = Generates Kate project files.
  Kate - NMake Makefiles       = Generates Kate project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Eclipse CDT4 - NMake Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - MinGW Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.

Error: could not load cache
Error: could not load cache

#### Build failed - see messages above.

### Include context on what you are trying to achieve

N/A

#### Context details

I am running it on windows 11, unreal engine 4.27.2.

### Include details of what you already did to find answers
I saw another solution which is to run the command prompt as adminstrator which I did, but I still get the same error as shown. 
Greatly appreciate any help given! Thank you!

MatvV04, 2023-04-08:
Hi, I had a similar issue, what worked for me was to manually install the most recent version of cmake 3.26.3 (when I'm typing this) while you are using version 3.14.7

Edmonds1997, 2025-08-28:
> Hi, I had a similar issue, what worked for me was to manually install the most recent version of cmake 3.26.3 (when I'm typing this) while you are using version 3.14.7

Update from 2025, getting the latest version was what I needed! There was another post that said to run Visual Studio Developer Command Prompt in admin mode. To anyone facing these issues, try both of those!