microsoft/AirSim #4825: ERROR: Unable to find plugin 'AirSim' (referenced via Blocks.uproject). Install it and try again, or remove it from the required plugin list.

DHrobotman, 2023-04-04:
Install UE4 (4.27) and AirSim 1.8.0 for the first time, then run the default environment in AirSim

`AirSim/Unreal/Environments/Blocks/Blocks.uproject`
Report the following error

```
Running /home/robocup/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project="/home/robocup/AirSim/Unreal/Environments/Blocks/Blocks.uproject" -TargetType=Editor -Progress -NoEngineChanges -NoHotReloadFromIDE

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine/Engine ~/UnrealEngine/Engine/Binaries/Linux
Using 'git status' to determine working set for adaptive non-unity build (/home/robocup/UnrealEngine).
Creating makefile for BlocksEditor (no existing makefile)
ERROR: Unable to find plugin 'AirSim' (referenced via Blocks.uproject). Install it and try again, or remove it from the required plugin list.
```

ai-naymul, 2023-04-12:
I think you are trying to run the default environment in `AirSim` using `Unreal Engine 4.27` and `AirSim 1.8.0.` However, you are encountering an error stating that the `AirSim `plugin cannot be found. This error suggests that the `AirSim `plugin is not installed or is not in the correct location. You may want to check if the `AirSim `plugin is installed and if it is in the correct location. If it is not installed, you can try installing it and then running the environment again.
:)

AthanasiosPetsanis, 2024-02-29:
> I think you are trying to run the default environment in `AirSim` using `Unreal Engine 4.27` and `AirSim 1.8.0.` However, you are encountering an error stating that the `AirSim `plugin cannot be [...]

Thank you genius... Everyone can read, don't state the obvious. The real question is why is the this problem occurs after following the steps of the documentation and how to check/install the airsim plugin properly..

ukd1, 2025-01-29:
Note, for those still trying to debug this - unless I missed it - I had the same error. Fixed by manually symlinking the built version of the plugin into the shared UE engine plugins folder. For me, this was:

```bash
% cd Unreal/Plugins
% ln -s AirSim /Users/Shared/Epic\ Games/UE_5.4/Engine/Plugins/
```

ukd1, 2025-01-29:
Note, for those still trying to debug this - unless I missed it - I had the same error. Fixed by manually symlinking the built version of the plugin into the shared UE engine plugins folder.

DavidKalajdzic, 2025-05-07:
@ukd1 can you explain where is the built version of the plugin?