microsoft/AirSim #4826: "Blocks could not be compile.Try rebuilding from source manually"

yingjiewx, 2023-04-14:
My Environennt:Airsim v1.2,Unreal4.18.3,Visual Studio2017
Due to the needs of running the project，I didn't use the latest version,
I have seen other discussions, but it seems that none of them can solve my problem.
When I was building Airsim, he popped up an error related to Eigen, which was manually downloaded and placed in/airlib/deps to resolve.
But when I tried to use plugin in other unreal projects, I found that it couldn't load at all, and even caused my project to not run.
So I was wondering if the testing environment for blocks could run, and then the above situation occurred.
And it is normal after I run update-from_git，but there is a display system that cannot find the specified file
[image]
The following are the issues that occurred when I ran upproject
[image]
[image]
Error reported when running blocks from vs
[image]
Can someone solve it？thanks！！

yingjiewx, 2023-04-14:
This error occurred on Win10

yingjiewx, 2023-04-15:
OH！I think I could end this issue now,The reason for this issue is due to the lack of the eigen library after the initial build.cmd.
It is an error to directly unzip the eigen library on the official website to airlib/deps/. 
Even if you successfully build to use airsim, you will still report errors similar to those in Figure 3.
 In fact, all of those errors are from the eigen file
The correct approach is：
1.Open the build.cmd with vs
2.Search for “eigen” keywords，you could find the powershell command like the following picture.
[image]
3.You will find that this link has become invalid and cannot download the eigen at all
4.And then You need to replace this link with the corresponding version of the official website's Eigen library link，
for example,If it is version 3.3.2, replace it with (https://gitlab.com/libeigen/eigen/-/archive/3.3.2/eigen-3.3.2.zip).
I have been stuck with this error for a long time, and I hope this issue can be helpful to those who later use the old version of Airsim！

closed 2023-04-15 by yingjiewx

Menes00, 2023-12-20:
I'm also using UE4.18.3 with VS2017 for a project, I get this error when I build AirSim, I don't quite understand how you solved it.

[image] 
@yingjiewx