microsoft/AirSim #4841: Airsim Clock Speed

Charles-Lim93, 2023-05-17:
## Question
### What's your question?

Hello. I'm using Airsim for Reinforcement Learning, and it is great that working so well. I'm wondering about ClockSpeed in the settings.json file. To speed up simulation I set 'ClockSpeed' as 5, which means 5 times faster.

It worked fine but I found out that the value from 5X ClockSpeed si (drone velocity, for example) is almost 1/5 compared to normal speed. I'm wondering whether that velocity value is depended to clockspeed. If it is true then the training network will get wrong information compare to what I expected. 

### Include context on what you are trying to achieve

Is it correct that airsim's value differs to clockspeed, and if it do so, then how we can (or make independent from Clock Speed)?

Thank you in advance.
#### Context details

### Include details of what you already did to find answers

wagh311, 2023-12-04:
Hello, I also encountered a similar problem to you. My task is to use DRL to train the drone to avoid obstacles, with ClockSpeed set to 5. When I complete the training, if I test with ClockSpeed set to 5, the drone can avoid collisions. But when I tested it with ClockSpeed set to 1, the drone crashed and seemed to forget the learned strategy. Have you solved the above related problems? Can you give me some advice? Thanks!

LucasJSch, 2023-12-04:
Hi @Charles-Lim93 @wagh311 . I'm interested in this issue.

Could you guys provide more info regarding the simulation scenario and how you measured the numbers you're providing here?

For a starter, the `settings.json` file would be a huge help. Also, in which branch/code version you're working?
Also, how you measured the speed of the drone when clockspeed is 5X? And what clocktype you are using? When I increment the clockspeed, I see also an increment in the drone speed, unlike what you're describing @Charles-Lim93 .

Last but not least: there's a chance that this problem is arising from your training scenario. E.g. the way you're feeding your algorithm the timestamps can be hardcoded to support just 5x clockspeed.

wagh311, 2023-12-05:
@Charles-Lim93 @LucasJSch 

Hello everyone, thank you for your replies. My current task is to use 3 drones to track 1 drone in the Block environment and achieve collision avoidance. The version of Airsim is 1.8.1. I just tried it and found that clockspeed will affect the real movement of the drone. Specifically, I set the target drone to fly along the Y-axis at a speed of 0.5m/s when clockspeed=5. However, when I used "getMultirotorState().kinematics_estimated.linear_velocity.y_val" to get the target's velocity, I found that it was only about 0.1m/s, which is 1/5 of the set value.

In my task, the action space of the drone performing the tracking task is set to “spaces.Box(low=np.array([-3.0, -3.0]), high=np.array([3.0, 3.0]),dtype=float)”, and the speed of the target drone to be tracked is 0.5m/s. So, for @Charles-Lim93 , 
 according to your suggestion, when I train with clockspeed=5, if I want ensure the RL agent can adapt to clock speed 5 to clock speed 1, should I multiply the action output by the policy network by 5 while setting the speed of the target drone to 2.5m/s? Do I need to multiply the action space by 5 as well?