microsoft/AirSim #4844: PX4 crashes the Unreal Engine 5.1.1 without any crash report (from UE)

srijanpal07, 2023-06-03:
## Bug report 

- AirSim Version/#commit: [Colosseum](https://github.com/CodexLabsLLC/Colosseum)
- UE/Unity version: UE 5.1.1
- autopilot version: v1.12.1
- OS Version: Windows 11/ wsl2 version: Ubuntu-18.04 LTS

### What's the issue you encountered?

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form
...
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative
ERROR [simulator] poll timeout 0, 0
ERROR [simulator] poll timeout 0, 0
ERROR [simulator] poll timeout 0, 0

**What's better than filing an issue? Filing a pull request :).**

doeschbj, 2024-02-01:
Hello, do you have any updates on this issue yet? I probably encountered a similar issue, where UE with AirSim crashes within 2 - 60 seconds after starting px4. The more drones I use the earlier it crashes and UE just gives me an access violation error to some 0x0...28 location.

srijanpal07, 2024-02-02:
> Hello, do you have any updates on this issue yet? I probably encountered a similar issue, where UE with AirSim crashes within 2 - 60 seconds after starting px4. The more drones I use the earlier it crashes and UE just gives me an access violation error to some 0x0...28 location.

I downgraded my system to Windows 10, because last time AirSim was supported, windows 11 did not came out and reinstalled unreal, wsl2, and everything from scratch in windows 10. It is working fine right now!

closed 2024-02-02 by srijanpal07