microsoft/AirSim #4846: Unreal Engine 4.27 no server for download.

joerowelll, 2023-06-07:
The Unreal Engine doesnt have version 4.27 to download, I get this error 
```
Updating dependencies:   0% (0/3525)...
Failed to download 'http://cdn.unrealengine.com/dependencies/2937058-0f6110266df14ff29898e518624a1ecc/086a185d97a278fdbe9fa922ecdfc94e4f08ea2c': The remote server returned an error: (403) Forbidden. (WebException)
```, 
upon inspection, the server is down, see https://forums.unrealengine.com/t/unable-to-build-4-27-from-source-setup-bat-403-forbidden-error/1156805 .
Which version of Unreal is reccommended otherwise?

cryptik, 2023-06-09:
I was able to download 4.27.2.  I did not see a 4.27... but 4.27.2 should work just as well.

kiririn39, 2023-07-29:
same problem, with 4.27.2

samiulextreem, 2023-08-22:
anyone have any solution? its very frustrating to come all the way and see that its not working.