microsoft/AirSim #4853: AirSim linux binary crashes

kanishkanarch, 2023-06-27:
## Bug report 

- AirSim Version/#commit: Binary tag 1.8.1
- UE/Unity version: N/A
- autopilot version: PX4-Autopilot commit `b874eaf5f90`
- OS Version: Ubuntu 20 Kernel 5.15.0-75-generic
- nVidia driver version: `nvidia-driver-525` (525.116.04-0ubuntu0.20.04.1 amd64)

### What's the issue you encountered?

AirSimNH.sh file crashes the binary pretty frequently, sometimes without even launching `make px4_sitl_default none_iris`, but more often when the px4 command is run. Some other issues discuss updating vulkan drivers, nvidia drivers, but to no avail.
[image]

### Settings

This is the [`settings.json`](https://gist.github.com/kanishkanarch/a2b1383b68290cfe71326942d2775122#file-settings-json) file that I'm using

### How can the issue be reproduced?

1. Run the downloaded binary `~/AirSimNH/LinuxNoEditor/AirSimNH.sh -ResX=700 -ResY=400 -windowed -settings"~/Documents/AirSim/settings.json"`
2. Run `make px4_sitl_default none_iris` from the autopilot folder (optional, as sometimes the above airsim binary crashes without even this)

### Include full error message in text form

Full output logs for: [airsim binary](https://gist.github.com/kanishkanarch/a2b1383b68290cfe71326942d2775122#file-airsim_binary_error_output) and for [px4](https://gist.github.com/kanishkanarch/a2b1383b68290cfe71326942d2775122#file-px4_error_output)

#### Text output

Relevant part of output according to me:
```
[2023.06.27-07.22.11:866][  0]LogVulkanRHI: Creating new VK swapchain with present mode 0, format 44, color space 0, num images 3
Connected to SITL over TCP.
Connecting to PX4 Control UDP port 14540, local IP 127.0.0.1, remote IP 127.0.0.1 ...
received first heartbeat
Ground control connected over UDP.
[2023.06.27-07.22.22:953][542]LogTemp: Connected to SITL over TCP.
[2023.06.27-07.22.22:953][542]LogTemp: Connecting to PX4 Control UDP port 14540, local IP 127.0.0.1, remote IP 127.0.0.1 ...
[2023.06.27-07.22.22:953][542]LogTemp: received first heartbeat
[2023.06.27-07.22.22:953][542]LogTemp: Ground control connected over UDP.
Got GPS lock
[2023.06.27-07.22.23:065][558]LogTemp: Got GPS lock
Preflight Fail: ekf2 missing data
[2023.06.27-07.22.23:954][680]LogTemp: Preflight Fail: ekf2 missing data
[logger] ./log/2023-06-27/07_22_23.ulg	
[2023.06.27-07.22.24:005][687]LogTemp: [logger] ./log/2023-06-27/07_22_23.ulg	
Got GPS Home Location
[2023.06.27-07.22.25:070][833]LogTemp: Got GPS Home Location
libc++abi: terminating with uncaught exception of type std::runtime_error: Error: parameter name 'COM_OBL_ACT' was not found
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
[2023.06.27-07.22.29:964][509]LogCore: === Critical error: ===
Unhandled Exception: SIGABRT: abort() called

[2023.06.27-07.22.29:964][509]LogCore: Fatal error!

0x00007fec2759600b libc.so.6!gsignal(+0xcb)
0x00007fec27575859 libc.so.6!abort(+0x12a)
0x0000000009c3cb06 AirSimNH!abort_message()
0x0000000009c3c9bf AirSimNH!demangling_terminate_handler()()
0x0000000009c3c873 AirSimNH!std::__terminate(void (*)())()
0x0000000009c3c0d6 AirSimNH!__cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*)()
0x0000000009c3c070 AirSimNH!__cxa_throw(+0x6f)
0x0000000008cfaaeb AirSimNH!mavlinkcom_impl::MavLinkNodeImpl::setParameter(mavlinkcom::MavLinkParameter)(+0x69a)
0x0000000008ceb7b0 AirSimNH!mavlinkcom::MavLinkNode::setParameter(mavlinkcom::MavLinkParameter)(+0x3f)
0x00000000033980e3 AirSimNH!msr::airlib::MavLinkMultirotorApi::sendParams() [/datadrive/UnrealEnvironments/AirSimEnvNH/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp:1552]
0x0000000003396245 AirSimNH!msr::airlib::MavLinkMultirotorApi::connectVehicle() [/datadrive/UnrealEnvironments/AirSimEnvNH/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp:1454]
0x000000000339582b AirSimNH!msr::airlib::MavLinkMultirotorApi::createMavEthernetConnection(msr::airlib::AirSimSettings::MavLinkConnectionInfo const&) [/datadrive/UnrealEnvironments/AirSimEnvNH/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp:1392]
0x000000000339343c AirSimNH!msr::airlib::MavLinkMultirotorApi::connect_thread() [/datadrive/UnrealEnvironments/AirSimEnvNH/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp:964]
0x0000000003399378 AirSimNH!void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (msr::airlib::MavLinkMultirotorApi::*)(), msr::airlib::MavLinkMultirotorApi*> >(void*) [/datadrive/ue_427/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/thread:291]
0x00007fec28017609 libpthread.so.0!UnknownFunction(0x8608)
0x00007fec27672133 libc.so.6!clone(+0x42)

[2023.06.27-07.22.29:976][509]LogExit: Executing StaticShutdownAfterError
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Malloc Size=102041 LargeMemoryPoolOffset=429993 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)```

CalvinSchwartz, 2023-10-03:
Happens also on the other precompiled environments for me.

kanishkanarch, 2023-10-04:
Hi Calvin, if you look at my error message closely, there's a `Error: parameter name 'COM_OBL_ACT' was not found`. If you're also having this error, please remove this parameter from your settings.json file and you're good to go.

varadVaidya, 2023-10-10:
> Hi Calvin, if you look at my error message closely, there's a `Error: parameter name 'COM_OBL_ACT' was not found`. If you're also having this error, please remove this parameter from your settings.json file and you're good to go.

I am also facing this same situation. This does fixes the problem, as PX4 will land, if the offboard computer doesn't send some messages at the rate of 2Hz. 
Removing this parameter, isn't really a ideal solution according to me.

kanishkanarch, 2023-10-10:
The parameter doesn't come with AirSim's binary, apparently. You can try using another parameter that goes by the name [`COM_OBL_RC_ACT`](https://docs.px4.io/v1.12/en/advanced_config/parameter_reference.html#COM_OBL_RC_ACT), but I think it will also produce the same error. The AirSim binaries (all of them) come undefined with the parameter `COM_OBL_ACT` and we can't do anything for the moment.
[image]

CalvinSchwartz, 2023-10-10:
At least on Linux. The AirSim binaries in Windows run perfectly fine with PX4.

My "solution" was to run AirSim on a Windows PC and publish all the PX4-stuff on the local IP Address/Port. Then PX4 running on my Linux machine accesses this data over the network. If anyone needs the configuration for this I can definitely share it tomorrow or so.

kanishkanarch, 2023-10-10:
That would be super helpful, please share as soon as possible!

kanishkanarch, 2023-10-11:
Thanks a lot for sharing!
Why did you do this, btw? Does this setup has any other advantages?

CalvinSchwartz, 2023-10-11:
> Thanks a lot for sharing!
> 
> Why did you do this, btw? Does this setup have any other advantages?

No, not really. It could have some performance benefits if you perform the simulation on a different machine than the Autopilot. But in my case it's simply to be able to run PX4 and AirSim without crashing. Personally I would prefer running AirSim on Linux too.