microsoft/AirSim #4860: Unable to build on MacOS with Xcode

santurini, 2023-07-12:
## Bug report 

- AirSim Version/#commit: [6688d27]
- UE/Unity version: UE 4.27
- autopilot version: N/A
- OS Version: MacOS Ventura 13.4.1

### What's the issue you encountered?

When trying to build *Blocks.xcworkspace* I get the following error:

[image]

### Settings

Default settings

### How can the issue be reproduced?

I've followed all the steps in [build_macos.md](https://microsoft.github.io/AirSim/build_macos/)

### Include full error message in text form

Cannot code sign because the target does not have an Info.plist file and one is not being generated automatically. Apply an Info.plist file to the target using the INFOPLIST_FILE build setting or generate one automatically by setting the GENERATE_INFOPLIST_FILE build setting to YES (recommended).

yrik, 2023-08-16:
I have the same issue...

Inglam, 2023-12-09:
Same issue. Any resolutions so far??

jordanhamel, 2024-03-06:
You can get past this issue, but I'm getting other build blockers around deprecated syntax and after I fix those, there were a ton of other issues in other files. I followed the mac build instructions but haven't gotten it working yet (M2 MBP). AirSim build fine still but stuck at building Blocks in Xcode.

martin-sn, 2024-11-13:
Any update on this?

AminoLEVA, 2025-02-25:
Also seeing this issue

DavidKalajdzic, 2025-04-18:
Same issue M4 MBP

apalvai, 2025-11-20:
Running into the same issue. Any update?