microsoft/AirSim #4863: roslaunch

LN-123-hub, 2023-07-21:
## Bug report 

- AirSim Version/#commit:
- UE/Unity version:
- autopilot version:
- OS Version:

### What's the issue you encountered?

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

LN-123-hub, 2023-07-21:
Because I was in a hurry to work on the project, I didn't debug in detail: the reason for the error was that simmode_name was an empty string, and it seemed that the Setting.json was not read successfully, but I have added Settings.json to the target path according to this plan
  https://github.com/microsoft/AirSim/issues/2009 Unfortunately it didn't work, so I forced the variable name directly in the Airlib part of the source code, and the program started. If you have other more fundamental solutions, please continue to leave a message

[image]

[image]

sandilyasg, 2023-11-18:
@LN-123-hub which settings.json do you use for this WSL setup? As in what is the location of the settings.json, is it in your windows wsl directory or in the Documents/Airsim folder on your WSL?