microsoft/AirSim #4866: When building the AirSim environment . /setup.sh, I get the following error: E: Package vulkan-utils not found is displayed

izakisyoutarou, 2023-07-28:
Sorry, I just finished setting up UnrealEngine and when I tried to set up AirSim, I got the following error.
This error occurred while setting up the environment
I'd like to know how to deal with it.
E: Package vulkan-utils not found.

sandilyasg, 2023-10-10:
@izakisyoutarou Try to install the mesa-vulkan-drivers package:
sudo apt install mesa-vulkan-drivers 
and 
sudo apt install vulkan-utils

Aditya12123, 2024-03-15:
I tried this. However, when I run sudo apt install valkan-utils, I still get: 
E: Package vulkan-utils not found
vulkaninfo does show the vulkan instance version:
VULKANINFO
==========
Vulkan Instance Version: 1.3.204

And yet I get that error E: Package vulkan-utils not found.
Can you please help me ?

Templarrr, 2024-07-05:
@izakisyoutarou  @Aditya12123 are you by chance on Ubuntu 22.04? 
I had the same problem, I fixed it by running
```
wget -qO- https://packages.lunarg.com/lunarg-signing-key-pub.asc | sudo tee /etc/apt/trusted.gpg.d/lunarg.asc
sudo wget -qO /etc/apt/sources.list.d/lunarg-vulkan-jammy.list http://packages.lunarg.com/vulkan/lunarg-vulkan-jammy.list
sudo apt update
```
before setup.sh
These commands are from the official Vulkan page:
https://vulkan.lunarg.com/doc/sdk/1.3.239.0/linux/getting_started_ubuntu.html

h2jaafar, 2024-07-05:
> @izakisyoutarou @Aditya12123 are you by chance on Ubuntu 22.04? I had the same problem, I fixed it by running
>
> ```
> wget -qO- https://packages.lunarg.com/lunarg-signing-key-pub.asc | sudo tee [...]

worked!

furkanunl11, 2024-10-16:
@Templarrr Worked bro thx.