microsoft/AirSim #4870: The imu in car mode has a bug

xuqingyuan2000, 2023-08-03:
## Bug report 

- AirSim Version/#commit:1.8.1
- UE/Unity version:4.27.2
- autopilot version:
- OS Version:ubuntu20.04

### What's the issue you encountered?

My system is ubuntu20.04. I use the scene in UE4.27.2 and airsim for simulation. I found that the IMU measurements have bug when using car mode , python api of car mode and ros together. For example, when the car is not moving on a flat surface, the Z-value of the linear acceleration in the imu measurement varies from 4 to 16.
Process:
(1) Set settings.json to car mode and equipped with IMU. The frequency of the IMU is 100Hz. The frequency of the camera image is 20Hz
(2) run:     AIRSIM/Unreal/Environment/Blocks/Blocks uproject
(3) In AIRSIM/ros, run:
roslaunch airsim_ros_pkgs airsim_node.launch
(4) In AIRSIM/PythonClient/car, run:
python3 hello_car.py
Note: Any Python program that controls car motion through the Python api can be run here
(5) rostopic echo  Check the published IMU topic data, it can be seen that when the car does not move, the Z-value of linear acceleration is wrong.

### Settings

settings.json:

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/main/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "ViewMode": "SpringArmChase",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "Sensors": {
        "Gps": {
          "SensorType": 3,
          "Enabled" : true
        },
        "Barometer": {
          "SensorType": 1,
          "Enabled" : true
        },
        "Magnetometer": {
          "SensorType": 4,
          "Enabled" : true
        },
        "Imu" : {
          "SensorType": 2,
          "Enabled": true
        },        
        "LidarCustom": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 64,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1.5,
          "DrawDebugPoints": false,
          "DataFrame": "SensorLocalFrame"
         }
      },
      "Cameras": {
        "front_left_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 100,
              "TargetGamma": 1.5
            },
            {
              "PublishToRos": 1,
              "ImageType": 1,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 100,
              "TargetGamma": 1.5
            }
          ],
          "X": 1.75, "Y": -0.06, "Z": -1.25,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "front_left_custom", "Visible": true}
  ]
}

### How can the issue be reproduced?

1.Set settings.json to car mode and equipped with IMU. The frequency of the IMU is 100Hz. The frequency of the camera image is 20Hz
2.run:     AIRSIM/Unreal/Environment/Blocks/Blocks uproject
3.In AIRSIM/ros, run:
roslaunch airsim_ros_pkgs airsim_node.launch
4. In AIRSIM/PythonClient/car, run:
python3 hello_car.py
Note: Any Python program that controls car motion through the Python api can be run here
5.  rostopic echo  Check the published IMU topic data, it can be seen that when the car does not move, the Z-value of linear acceleration is wrong.
### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

szydgithub, 2024-12-27:
I'm currently having an issue with the IMU, my IMU timestamps are duplicating, do you have this issue at the moment？

szydgithub, 2024-12-27:
My OS Version is Unbuntu 18.04

ssaigarimella, 2025-07-04:
@xuqingyuan2000 @szydgithub any updates on how you may have fixed this? thanks