microsoft/AirSim #4888: roslaunch airsim_ros_pkgs airsim_node.launch;' is stuck at 'process[ned_to_enu_pub-3]: started with pid [1192]'

bb6698, 2023-09-26:
Hi! I use ros_wrapper in WSL2 it had installed Ubuntu 20.04 and ROS noetic. 

The problem is : roslaunch airsim_ros_pkgs airsim_node.launch output:=screen
... logging to /home/bvillalb/.ros/log/95579acc-5c55-11ee-8885-45dabb331bc6/roslaunch-DESKTOP-526MF71-1117.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://DESKTOP-526MF71:42671/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: localhost
 * /airsim_node/is_vulkan: False
 * /airsim_node/publish_clock: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.01
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [1125]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 95579acc-5c55-11ee-8885-45dabb331bc6
process[rosout-1]: started with pid [1135]
started core service [/rosout]
process[airsim_node-2]: started with pid [1142]
process[ned_to_enu_pub-3]: started with pid [1143]
Waiting for connection - XXXXX

I tried an installation: pip3 install msgpack-rpc-python and then pip3 install airsim, but I still have the same problem.

I need your help!

Regards

bb6698, 2023-09-26:
I find the error :)

closed 2023-09-26 by bb6698

LeightonWang, 2024-03-16:
Hello, I'm running into this problem as well. Could you please share how you've solve it?