microsoft/AirSim #4890: Why does the drone lose altitude when commanding x- and y-axis velocities?

bb6698, 2023-10-02:
Hi! I'm using ROS wrapper in Airsim. I execute launch airsim_node.launch and I want to commanding velocities in exe x or exe y using topic /airsim_node/drone/vel_cmd_world_frame or topic /airsim_node/drone/vel_cmd_body_frame, but when I do it the drone lose altitude in exe z ultil the drone end in the ground.

Why is this happening?

liewyungjun, 2023-10-15:
hi I've also faced this same issue when issuing commands using /airsim_node/drone/vel_cmd_world_frame. I dont have an answer but if you could work with position, the PID controller node helps with this issue.

MehdiShahbazi, 2023-12-15:
I don't know why it's happening but use the following code you can pitch in x and y directions without losing altitude:
```
self.client.moveByVelocityZBodyFrameAsync(
    vx = 0, 
    vy = 0,
    z  = max_altitude, 
    duration = 1, 
    yaw_mode = airsim.YawMode(is_rate=True, yaw_or_rate=0)
    ).join()
```

sandilyasg, 2024-02-03:
@liewyungjun liewyungjun can you please share what you mean by PID controller node? Are you referring to the controller gain api calls?