microsoft/AirSim #4892: Package 'clang-8' has no installation candidate, Ubuntu 22.04

iagupov, 2023-10-11:
## Bug report 

- AirSim Version/#commit: 1.8.1
- UE/Unity version: 4.27
- autopilot version: ArduPilot 4.4.1
- OS Version: Ubuntu 22.04.3
- python 3.9.17

### What's the issue you encountered?

I follow the instruction: https://microsoft.github.io/AirSim/build_linux/ to install AirSim in Ubuntu
In a section "Build AirSim" I cloned repository and then executed `./setup.sh`
This provoke an error: "Package 'clang-8' has no installation candidate"
I did not find any similar issue or any relative information. The full log is shown below. Please help if you have any idea how to fix it. I saw that clang is not supported for Ubuntu 22.04, but maybe there is a way to run it without reinstalling Ubuntu.

### Settings

### How can the issue be reproduced?

1. Just follow the AirSim installation procedure from https://microsoft.github.io/AirSim/build_linux/

### Include full error message in text form
./setup.sh

> user@MSK-PC-0726:~/cv_dev/ardupilot/AirSim$ ./setup.sh 
> + set -e
> +++ dirname ./setup.sh
> ++ cd .
> ++ pwd
> + SCRIPT_DIR=/home/user/cv_dev/ardupilot/AirSim
> + pushd /home/user/cv_dev/ardupilot/AirSim
> + downloadHighPolySuv=true
> + MIN_CMAKE_VERSION=3.10.0
> ++ uname
> + '[' Linux == Darwin ']'
> + DEBUG=false
> + [[ 0 -gt 0 ]]
> ++ uname
> + '[' Linux == Darwin ']'
> + sudo apt-get update
> Hit:1 http://archive.ubuntu.com/ubuntu jammy InRelease
> Hit:2 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
> Hit:3 http://archive.ubuntu.com/ubuntu jammy-security InRelease                                          
> Hit:4 http://archive.ubuntu.com/ubuntu jammy-backports InRelease                                         
> Hit:5 http://archive.canonical.com/ubuntu jammy InRelease                                                
> Reading package lists... Done            
> + sudo apt-get -y install --no-install-recommends lsb-release rsync software-properties-common wget libvulkan1 vulkan-utils
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done         
> Note, selecting 'vulkan-tools' instead of 'vulkan-utils'
> lsb-release is already the newest version (11.1.0ubuntu4).
> wget is already the newest version (1.21.2-2ubuntu1).
> rsync is already the newest version (3.2.7-0ubuntu0.22.04.2).
> software-properties-common is already the newest version (0.99.22.7).
> libvulkan1 is already the newest version (1.3.261.0~rc1-1lunarg20.04-1).
> vulkan-tools is already the newest version (1.3.261.0~rc2-1lunarg20.04-1).
> The following packages were automatically installed and are no longer required:
>   amd64-microcode binutils-arm-linux-gnueabihf cpp-11-arm-linux-gnueabihf cpp-arm-linux-gnueabihf cuda-command-line-tools-11-2 cuda-compiler-11-2 cuda-cudart-11-2 cuda-cudart-dev-11-2
>   cuda-cuobjdump-11-2 cuda-cupti-11-2 cuda-cupti-dev-11-2 cuda-cuxxfilt-11-2 cuda-documentation-11-2 cuda-driver-dev-11-2 cuda-gdb-11-2 cuda-libraries-11-2 cuda-libraries-dev-11-2
>   cuda-memcheck-11-2 cuda-nsight-11-2 cuda-nsight-compute-11-2 cuda-nsight-systems-11-2 cuda-nvcc-11-2 cuda-nvdisasm-11-2 cuda-nvml-dev-11-2 cuda-nvprof-11-2 cuda-nvprune-11-2 cuda-nvrtc-11-2
>   cuda-nvrtc-dev-11-2 cuda-nvtx-11-2 cuda-nvvp-11-2 cuda-samples-11-2 cuda-sanitizer-11-2 cuda-toolkit-11-2 cuda-tools-11-2 cuda-visual-tools-11-2 dctrl-tools dkms
>   gcc-11-arm-linux-gnueabihf-base gcc-11-cross-base gcc-12-cross-base intel-microcode iucode-tool libasan6-armhf-cross libatomic1:i386 libatomic1-armhf-cross libc6-armhf-cross
>   libc6-dev-armhf-cross libcublas-11-2 libcublas-dev-11-2 libcufft-11-2 libcufft-dev-11-2 libcurand-11-2 libcurand-dev-11-2 libcusolver-11-2 libcusolver-dev-11-2 libcusparse-11-2
>   libcusparse-dev-11-2 libdrm-amdgpu1:i386 libdrm-intel1:i386 libdrm-nouveau2:i386 libdrm-radeon1:i386 libdrm2:i386 libegl-mesa0:i386 libegl1:i386 libelf1:i386 libexpat1:i386 libgbm1:i386
>   libgcc-11-dev-armhf-cross libgcc-s1-armhf-cross libgdk-pixbuf-xlib-2.0-0 libgdk-pixbuf2.0-0 libgl1:i386 libgl1-mesa-dri:i386 libglapi-mesa:i386 libgles2:i386 libglvnd0:i386 libglx-mesa0:i386
>   libglx0:i386 libgomp1-armhf-cross libicu70:i386 libllvm15:i386 libnpp-11-2 libnpp-dev-11-2 libnvidia-cfg1-535 libnvidia-common-535 libnvidia-compute-530:i386 libnvidia-compute-535:i386
>   libnvidia-decode-530:i386 libnvidia-decode-535 libnvidia-decode-535:i386 libnvidia-encode-530:i386 libnvidia-encode-535 libnvidia-encode-535:i386 libnvidia-extra-535 libnvidia-fbc1-530:i386
>   libnvidia-fbc1-535 libnvidia-fbc1-535:i386 libnvidia-gl-535:i386 libnvjpeg-11-2 libnvjpeg-dev-11-2 libomxil-bellagio-bin libomxil-bellagio0 libopengl0:i386 libpciaccess0:i386
>   libsensors5:i386 libstdc++-11-dev-armhf-cross libstdc++6:i386 libstdc++6-armhf-cross libubsan1-armhf-cross libva2:i386 libvdpau1:i386 libx11-xcb1:i386 libxcb-dri2-0:i386 libxcb-dri3-0:i386
>   libxcb-glx0:i386 libxcb-present0:i386 libxcb-randr0:i386 libxcb-shm0:i386 libxcb-sync1:i386 libxcb-xfixes0:i386 libxfixes3:i386 libxml2:i386 libxshmfence1:i386 libxxf86vm1:i386
>   linux-libc-dev-armhf-cross mesa-utils-bin mesa-vdpau-drivers:i386 nsight-compute-2020.3.1 nsight-systems-2020.4.3 nvidia-compute-utils-535 nvidia-firmware-535-535.113.01
>   nvidia-kernel-common-535 nvidia-kernel-source-535 nvidia-modprobe nvidia-prime nvidia-settings nvidia-utils-535 screen-resolution-extra thermald usb-modeswitch usb-modeswitch-data
>   vdpau-driver-all:i386 xserver-xorg-video-nvidia-535 xul-ext-ubufox
> Use 'sudo apt autoremove' to remove them.
> 0 upgraded, 0 newly installed, 0 to remove, and 3 not upgraded.
> ++ lsb_release -rs
> ++ cut -d. -f1
> + VERSION=22
> + '[' 22 -lt 17 ']'
> + sudo apt-get install -y clang-8 clang++-8 libc++-8-dev libc++abi-8-dev
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done         
> Note, selecting 'librust-clang-sys-1.3+clang-8-0-dev' for regex 'clang++-8'
> Note, selecting 'clang-8' for regex 'clang++-8'
> Note, selecting 'librust-clang-sys+clang-8-0-dev' for regex 'clang++-8'
> Note, selecting 'librust-clang-sys-1+clang-8-0-dev' for regex 'clang++-8'
> Note, selecting 'librust-clang-sys-1.3.0+clang-8-0-dev' for regex 'clang++-8'
> Note, selecting 'librust-clang-sys-dev' for regex 'librust-clang-sys+clang-8-0-dev'
> Note, selecting 'librust-clang-sys-dev' for regex 'librust-clang-sys-1+clang-8-0-dev'
> Note, selecting 'librust-clang-sys-dev' for regex 'librust-clang-sys-1.3+clang-8-0-dev'
> Note, selecting 'librust-clang-sys-dev' for regex 'librust-clang-sys-1.3.0+clang-8-0-dev'
> Package 'clang-8' is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or is only available from another source
> 
> E: Package 'clang-8' has no installation candidate
> E: Unable to locate package libc++-8-dev
> E: Unable to locate package libc++abi-8-dev
>

AgafonovAero, 2023-10-13:
Some problem!!!

chenfei8888, 2023-10-15:
I installed clang-11 and changed it in setup.sh, but build.sh failed to find -lc++fs: no file or directory

rishi1906, 2023-10-19:
facing the same issue, has someone found the solution? It looks like clang-8 is not supported in Ubuntu 22.04. @chenfei8888 did you figure out the solution?

Kanaderu, 2023-10-26:
According to to this [SO post](https://stackoverflow.com/questions/74891427/make-error-library-not-found-for-lstdcfs) it seems like the C/C++ filesystem library was added to the standard library so removing them should suffice.

Changes to remove them here:
https://github.com/Kanaderu/AirSim/commit/ac76e3908a768cc6e6fada54b77897695f48bcb7

Seems to build successfully, haven't tested anything as of yet.

JulianDev24, 2023-11-07:
I installed it with clang-12 here is what I have changed within the files. Im using it with Unity

`` Row | Old Value --> New Value ``

setup.sh
``
53 | vulkan-utils --> vulkan-tools
63 | sudo apt-get install -y clang-8 clang++-8 libc++-8-dev libc++abi-8-dev --> sudo apt-get install -y clang-12 clang++-12 libc++-12-dev libc++abi-12-dev
``

build.sh
``
67 | export CC="gcc-8" --> export CC="gcc-12"
68 | export CXX="g++-8" --> export CXX="g++-12"
70 | export CC="clang-8" --> export CC="clang-12"
71 | export CXX="clang++-12" --> export CXX="clang++-12"
``

cmake/cmake-modules/CommonSetup.cmake
``
62 | set(CXX_EXP_LIB "-L${LLVM_LIBRARY_DIRS} -lc++fs -ferror-limit=10") --> set(CXX_EXP_LIB "-L${LLVM_LIBRARY_DIRS} -ferror-limit=10")
64 | set(CXX_EXP_LIB "-lstdc++fs -fmax-errors=10 -Wnoexcept -Wstrict-null-sentinel") --> set(CXX_EXP_LIB "-fmax-errors=10 -Wnoexcept -Wstrict-null-sentinel")
``

ros2/src/airsim_ros_pks/CMakeLists.txt
``
37 | -lstdc++fs -fmax-errors=10") --> -fmax-errors=10")
``

Unity/build.sh
``
21 | export CC=/usr/local/opt/llvm@8/bin/clang --> export CC=/usr/local/opt/llvm@12/bin/clang
22 | export CXX=/usr/local/opt/llvm@8/bin/clang++ --> export CXX=/usr/local/opt/llvm@12/bin/clang++
24 | export CC="clang-8" --> export CC="clang-12"
25 | export CXX="clang++-8" --> export CXX="clang++-12"
``

rishi1906, 2023-11-25:
> I installed it with clang-12 here is what I have changed within the files. Im using it with Unity
>
> `Row | Old Value --> New Value`
>
> setup.sh `53 | vulkan-utils --> vulkan-tools 63 | sudo apt-get [...]

I got following errors, -lstdc++ No such file or directory 

```
> ./build.sh
+ debug=false
+ gcc=false
+ [[ 0 -gt 0 ]]
+ RPC_VERSION_FOLDER=rpclib-2.3.0
+ '[' '!' -d ./external/rpclib/rpclib-2.3.0 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Linux == Darwin ']'
+ false
+ export CC=clang-12
+ CC=clang-12
+ export CXX=clang++-12
+ CXX=clang++-12
+ [[ ! -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ CMAKE_VARS=
++ uname
+ '[' Linux == Darwin ']'
+ pushd build_release
+ false
+ folder_name=Release
+ /usr/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 12.0.1
-- The CXX compiler identification is Clang 12.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang-12 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /usr/bin/clang++-12
-- Check for working CXX compiler: /usr/bin/clang++-12 - broken
CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCXXCompiler.cmake:62 (message):
  The C++ compiler

    "/usr/bin/clang++-12"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/iit/AirSim/build_release/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_ba24f/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_ba24f.dir/build.make CMakeFiles/cmTC_ba24f.dir/build
    gmake[1]: Entering directory '/home/iit/AirSim/build_release/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_ba24f.dir/testCXXCompiler.cxx.o
    /usr/bin/clang++-12    -MD -MT CMakeFiles/cmTC_ba24f.dir/testCXXCompiler.cxx.o -MF CMakeFiles/cmTC_ba24f.dir/testCXXCompiler.cxx.o.d -o CMakeFiles/cmTC_ba24f.dir/testCXXCompiler.cxx.o -c /home/iit/AirSim/build_release/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_ba24f
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_ba24f.dir/link.txt --verbose=1
    /usr/bin/clang++-12 CMakeFiles/cmTC_ba24f.dir/testCXXCompiler.cxx.o -o cmTC_ba24f 
    /usr/bin/ld: cannot find -lstdc++: No such file or directory
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    gmake[1]: *** [CMakeFiles/cmTC_ba24f.dir/build.make:100: cmTC_ba24f] Error 1
    gmake[1]: Leaving directory '/home/iit/AirSim/build_release/CMakeFiles/CMakeTmp'
    gmake: *** [Makefile:127: cmTC_ba24f/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/iit/AirSim/build_release/CMakeFiles/CMakeOutput.log".
See also "/home/iit/AirSim/build_release/CMakeFiles/CMakeError.log".
+ popd
~/AirSim ~/AirSim
+ rm -r build_release
+ exit 1
```

rishi1906, 2023-11-25:
> I got following errors, -lstdc++ No such file or directory
>
> ```
> + debug=false
> + gcc=false
> + [[ 0 -gt 0 ]]
> + RPC_VERSION_FOLDER=rpclib-2.3.0
> + '[' '!' -d ./external/rpclib/rpclib-2.3.0 ']'
> + '[' -d [...]

Building with -gcc flag solved the issue

https://github.com/microsoft/AirSim/issues/4507#issuecomment-1311169211

chenfei8888, 2023-11-28:
> facing the same issue, has someone found the solution? It looks like clang-8 is not supported in Ubuntu 22.04. @chenfei8888 did you figure out the solution?

I ended up using ubuntu20.04.

mikeb127, 2024-01-03:
> I got following errors, -lstdc++ No such file or directory
>
> ```
> + debug=false
> + gcc=false
> + [[ 0 -gt 0 ]]
> + RPC_VERSION_FOLDER=rpclib-2.3.0
> + '[' '!' -d ./external/rpclib/rpclib-2.3.0 ']'
> + '[' -d [...]

Just in case anyone else arrives here via google. Run the following in terminal and then re-run build.sh:

sudo apt-get -y install libunwind-dev libstdc++-12-dev

Should work now.

Happy to create a PR for @JulianDev24 + my changes if repo maintainers are interested?

ben-xD, 2024-03-04:
> Happy to create a PR for @JulianDev24 + my changes if repo maintainers are interested?

The maintainers were unfortunately laid off. Initially, microsoft announced they were deprecated this repo for a newer, "Project Airsim" (probably an internal reorg). In October 2023, https://wonderfulengineering.com/microsoft-just-shut-down-its-industrial-metaverse-project-airsim-and-laid-off-its-team/
> Microsoft Just Shut Down Its ‘Industrial Metaverse’ Project Airsim And Laid Off Its Team

I wonder if someone should/could fork it, maybe that person could be in many months time - just getting started with Airsim right now 😂️

xxEoD2242, 2024-03-04:
@ben-xD We did: [Colosseum](https://github.com/CodexLabsLLC/Colosseum)

ben-xD, 2024-03-05:
@xxEoD2242, your repo has the same issues since it still uses clang 8 and references outdated libraries like `vulkan-utils`?

https://github.com/CodexLabsLLC/Colosseum/blob/2383b0adfc674d34e68cb05ef163ca1c00479352/setup.sh#L53

https://github.com/CodexLabsLLC/Colosseum/blob/2383b0adfc674d34e68cb05ef163ca1c00479352/setup.sh#L63

wouter-heerwegh, 2024-03-06:
We're struggling with a similar issue on our [sim](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator). Changing clang version from 8 &rarr; 12 breaks compatibility on Ubuntu 20.04, because of pthread symbols that are exposed in the rpclib library that are undefined for the cross-compile tool for Unreal Engine 4.27. Apparently [this PR](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/pull/359) fixes the issues on Ubuntu 22.04. We do however use an older version of Airsim that was stripped down a lot.

@xxEoD2242 Do you see a similar issue when changing from clang version 8 &rarr; 12 when using UE 5+? The symbol I'm talking about is `pthread_cond_clockwait` which is available since glibc 2.30, while the crosscompile toolchain for UE 4.27 uses glibc v2.17.

lu-chenxi, 2024-06-18:
> I installed it with clang-12 here is what I have changed within the files. Im using it with Unity
>
> `Row | Old Value --> New Value`
>
> setup.sh `53 | vulkan-utils --> vulkan-tools 63 | sudo apt-get [...]

it works for me!
thx!

nabeelsherazi, 2024-09-02:
Similar to @ben-xD adding that I was also able to get this to work on Ubuntu 24.04.1 by replacing references to clang-8 with clang-18 (the version of clang you get with `sudo apt install clang`) and removing linker references to `-lc++fs`. Also made the vulkan-tools change, but no need to do the gcc change, since I wasn't building with gcc anyway (didn't even try it tbh).

In other interesting news, the LinuxBlocks 1.8.1 binary works again -- I guess they got Vulkan figured out. Woo!

limathub, 2025-03-17:
> Similar to [@ben-xD](https://github.com/ben-xD) adding that I was also able to get this to work on Ubuntu 24.04.1 by replacing references to clang-8 with clang-18 (the version of clang you get with [...]

@nabeelsherazi Were you able to open the Blocks unreal project in Unreal 4.27? (just like it was mentioned in a comment of @ben-xD)