microsoft/AirSim #4920: Not able to install airsim python

adityaomar3, 2024-01-28:
## Bug report 

- AirSim Version/#commit:
- UE/Unity version:
- autopilot version:
- OS Version:

### What's the issue you encountered?

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form
 pip install airsim 
Collecting airsim
  Using cached airsim-1.8.1.tar.gz (20 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [27 lines of output]
      Traceback (most recent call last):
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main    
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Omar\AppData\Local\Temp\pip-build-env-p10yc3u8\overlay\Lib\site-packages\setuptools\build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Omar\AppData\Local\Temp\pip-build-env-p10yc3u8\overlay\Lib\site-packages\setuptools\build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "C:\Users\Omar\AppData\Local\Temp\pip-build-env-p10yc3u8\overlay\Lib\site-packages\setuptools\build_meta.py", line 480, in run_setup
          super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
        File "C:\Users\Omar\AppData\Local\Temp\pip-build-env-p10yc3u8\overlay\Lib\site-packages\setuptools\build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 2, in <module>
        File "C:\Users\Omar\AppData\Local\Temp\pip-install-7xlbe5zy\airsim_52dcecb4260c4b7aa1dbabc16a8c5fea\airsim\__init__.py", line 1, in <module>
          from .client import *
        File "C:\Users\Omar\AppData\Local\Temp\pip-install-7xlbe5zy\airsim_52dcecb4260c4b7aa1dbabc16a8c5fea\airsim\client.py", line 3, in <module>
          from .utils import *
        File "C:\Users\Omar\AppData\Local\Temp\pip-install-7xlbe5zy\airsim_52dcecb4260c4b7aa1dbabc16a8c5fea\airsim\utils.py", line 1, in <module>
          import numpy as np #pip install numpy
          ^^^^^^^^^^^^^^^^^^
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

[notice] A new release of pip is available: 23.2.1 -> 23.3.2
[notice] To update, run: python.exe -m pip install --upgrade pip
PS C:\project\photogrammetry> pip install airsim 
Collecting airsim
  Using cached airsim-1.8.1.tar.gz (20 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [27 lines of output]
      Traceback (most recent call last):
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Omar\AppData\Local\Temp\pip-build-env-gfvytytx\overlay\Lib\site-packages\setuptools\build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Omar\AppData\Local\Temp\pip-build-env-gfvytytx\overlay\Lib\site-packages\setuptools\build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "C:\Users\Omar\AppData\Local\Temp\pip-build-env-gfvytytx\overlay\Lib\site-packages\setuptools\build_meta.py", line 480, in run_setup
          super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
        File "C:\Users\Omar\AppData\Local\Temp\pip-build-env-gfvytytx\overlay\Lib\site-packages\setuptools\build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 2, in <module>
        File "C:\Users\Omar\AppData\Local\Temp\pip-install-4likky2c\airsim_997319d56f7d48fca1814ce8fe1b6524\airsim\__init__.py", line 1, in <module>
          from .client import *
        File "C:\Users\Omar\AppData\Local\Temp\pip-install-4likky2c\airsim_997319d56f7d48fca1814ce8fe1b6524\airsim\client.py", line 3, in <module>
          from .utils import *
        File "C:\Users\Omar\AppData\Local\Temp\pip-install-4likky2c\airsim_997319d56f7d48fca1814ce8fe1b6524\airsim\utils.py", line 1, in <module>
          import numpy as np #pip install numpy
          ^^^^^^^^^^^^^^^^^^
      ModuleNotFoundError: No module named 'numpy'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

**What's better than filing an issue? Filing a pull request :).**

Augustiu, 2024-02-06:
i met the same problem,how to fix it?

Mohmansy94, 2024-02-07:
Try to use python version 3.8, it should work without any problem.

Garans, 2024-03-11:
I used version 3.8 but met the same problem, for solved this issue I patched setup.py

durach, 2024-03-13:
@Garans, thanks, that helped. But I experience the issue only in `pyenv` (Mac). Everything worked fine on `conda` (Win)

Garans, 2024-03-13:
I'm using this lib with remote connect from Mac to win

MiguelPujol, 2024-04-10:
I have the same issue when trying to install airsim python package version 1.6.0, 1.7.0 and 1.8.1 (latest). 
However I did manage to successfully install version 1.5.0.

henryhellmann, 2024-10-23:
I am experiencing this problem as well

RuizhiRuan, 2024-11-25:
It seems to work:
pip install numpy
pip install airsim