microsoft/AirSim #4922: Building AirSim on Mac OS Sonoma with Apple Silicon chip

NikitaYeryomin, 2024-02-01:
## Question
### What's your question?
I was previously able to successfully build AirSim Blocks environment on MacOS BigSur with intel chip. However, getting issues building it on new Mac with Mac OS Sonoma 14.1 and Xcode 15.2 (please see attached screenshot). There are a bunch of deprecation warnings/errors  which can't be removed, e.g. 
IsTriviallyCopyAssignable.h:13:17 builtin __has_trivial_assign is deprecated; use __is_trivially_assignable instead [-Werror,-Wdeprecated-builtins]

I tried to enable the -Wno-deprecated-builtins flag as shown on the screenshot, but it still does not help.
Also tried to follow the workaround described here: https://forums.unrealengine.com/t/wdeprecated-builtins-error-when-trying-to-build-ue4-on-xcode-15-0-1/1404880/3
But Xcode 13.4.1 is crashing when launched on Sonoma unfortuntely.

Kindly looking for an advice. 

### Include context on what you are trying to achieve

[image]

#### Context details

### Include details of what you already did to find answers

jeanvins, 2024-02-06:
Same issue here... seems like we need Xcode 13 for building the engine. 
I tried to build with Xcode 14 with Project Format "Xcode 13 - compatible" but I got the same problem. 

Did you find a solution?

NikitaYeryomin, 2024-02-07:
> Same issue here... seems like we need Xcode 13 for building the engine. I tried to build with Xcode 14 with Project Format "Xcode 13 - compatible" but I got the same problem.
> 
> Did you find a solution?

Nope, switched to my old mac for now, but definitely it's a huge issue.

juan5705, 2024-02-09:
Ok, on my side I was finally able to build Blocks environment on Sonoma using Colosseum and XCode 14.3.1.

nikita-noteunlimited, 2024-02-10:
> Ok, on my side I was finally able to build Blocks environment on Sonoma using Colosseum and XCode 14.3.1.

Great, can you share some details?

greglo, 2024-03-01:
@nikita-noteunlimited I just got Colosseum running on a M1/Sonoma/XCode 15.2 with the following steps

```
1. Fresh install of Colosseum and Unreal 5.2 (set as current)
2. In Colosseum, update setup.sh and build.sh to use llvm 13 (not 8)
3. Update setup.sh to add `\-DCMAKE_OSX_ARCHITECTURES=x86_64` to CMAKE_FLAGS
4. After opening the XCode Workspace, in Blocks and blocks_build > Build Settings > Search for arm64
  - Update all values to x86_64
5. XCode Top menu > Product > Destinations > Destination Architectures > Enable "Show both"
6. Build using "My Mac (Rosetta)"
```

AYHSASIDDIKADOLA, 2024-06-12:
> @nikita-noteunlimited I just got Colosseum running on a M1/Sonoma/XCode 15.2 with the following steps
Can you please explain more point no 3? any tutorial please ?im woring for my research im stuck here

nikita-noteunlimited, 2024-06-14:
> > @nikita-noteunlimited I just got Colosseum running on a M1/Sonoma/XCode 15.2 with the following steps
> > Can you please explain more point no 3? any tutorial please ?im woring for my research im stuck here

@AYHSASIDDIKADOLA I believe your question is for @greglo, since I have not tried the approach he described above.

enterpolation, 2024-07-01:
Anyone tried on M3? Does it work?

greglo, 2024-07-01:
@nikita-noteunlimited See the video [here](https://www.youtube.com/watch?v=hXYWLF3QsNk)

I may have been looking at a different version of the scripts when I wrote the above since you're right, it doesn't appear necessary in setup. The appropriate flag is added in `build.sh` already: https://github.com/microsoft/AirSim/blob/main/build.sh#L100

@enterpolation I'm afraid I don't have an M3 system that I can test it on. I don't see any reason why that error would be related to having an M3 processor but anything's possible

Chop91, 2024-07-07:
> @nikita-noteunlimited See the video [here](https://www.youtube.com/watch?v=hXYWLF3QsNk)
>
> I may have been looking at a different version of the scripts when I wrote the above since you're right, it [...]

@greglo Somehow they removed those setting entirely from build.sh: https://github.com/CodexLabsLLC/Colosseum/blob/main/build.sh

I am struggeling since hours to get this running on my Mac Studio M2 with Sonoma 14.3 and XCode 15.4

AIM-Focus-Dev, 2025-03-02:
> Anyone tried on M3? Does it work? For me the following algorithm fails on the step
> 
> `./GenerateProjectFiles.sh /Users/Shared/Epic\ Games/UE_5.2`
> 
> with an error
> 
> `Unhandled exception: System.IO.DirectoryNotFoundException: Could not find a part of the path '/Users/interpolation/Colosseum/Unreal/Environments/Blocks/Intermediate/Build/BuildRules/BlocksModuleRulesManifest.json'.`

@enterpolation Not sure if you got it, but for anyone else for me the named was saved differently as:

`./GenerateProjectFiles.sh /Users/Shared/Epic Games/UE_5.2`

Rather than

`./GenerateProjectFiles.sh /Users/Shared/Epic\ Games/UE_5.2`

har5h1l, 2025-05-10:
Hey, I'm not up to date on a lot of this stuff, but I'm trying to get AirSim working on MacOS Sequoia. Following the official instructions and am getting 19 issues on Xcode. Tried change Xcode versions didn't work. It seems like Unreal Engine being on 4.27 and MacOS being updated and Xcode have a lot of inconsistencies. If anyone can help me out or refer me to anything on what I could do please let me know. I'm trying to get AirSim set up to use it with the API.