microsoft/AirSim #4941: AirSim

detront, 2024-03-13:
## Bug report 

- AirSim Version/#commit:
- UE/Unity version:
- autopilot version:
- OS Version:

### What's the issue you encountered?

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

mikolaj1981, 2024-03-20:
HI I installed airsim 1.8.1  in console i try open its not working the same in visual studio code i do somthing do now is not working in console before did work but visual studio not working but in visual studio code in terminal was working

mikolaj1981, 2024-03-20:
when i write command pip install airsim that  me show airsim its installed and show version 1.8.1 i have python 3.8  i try make program in visual studio but is not work

reza-shahriari, 2024-07-21:
HI,
check interpereter of VS. if you can see the airsim in terminal but not in VS it means they are using different environments.
[this linke](https://learn.microsoft.com/en-us/visualstudio/python/managing-python-environments-in-visual-studio?view=vs-2022) could be helpful