microsoft/AirSim #4942: more than 500 multi-agents simulation at one time

ZhangShaungnj, 2024-03-14:
How can I simulate more than 500 multi-rotors at one time.
**Here is my  computer hardware list：**
                                                           cpu：AMD EPYC 9654 96-Core Processor            
                                                           ram：128g
                                                           gpu：4090
I find the simulation will not display smoothly in UE everytime the number over 70~80
I have tried join(),multi-process,threading.They are not work

ZhangShaungnj, 2024-03-14:
[image]
[image]
[image]

Midas75, 2024-05-31:
AirSim adopts a spin wait similar to sleep (0) in some low-level sleep implementations, which causes threads to be repeatedly scheduled on the CPU core when multiple drones are simulated simultaneously, resulting in high kernel state consumption. When the number of simulated drones approaches the number of CPU cores/CPU threads, performance will plummet, so it can be seen that in your configuration, 80 drones correspond to your CPU96 core. The solution is to replace its internal spin implementation (approximately four times) with a plain sleep implementation and recompile AirLib for use in the Unreal project. Using this method, I can achieve 400+drone simulation based on i7-12700 in Unity scenarios.

Coinstine, 2025-04-24:
> AirSim adopts a spin wait similar to sleep (0) in some low-level sleep implementations, which causes threads to be repeatedly scheduled on the CPU core when multiple drones are simulated [...]

hi dude, could u tell me the detail how u replace the internal spin implementation with a plain sleep one? thank u sooooo much!!

Midas75, 2025-04-24:
> hi dude, could u tell me the detail how u replace the internal spin implementation with a plain sleep one? thank u sooooo much!!

Hi,
I forked AirSim and applied the implementation, which you can find [here](https://github.com/Midas75/AirSim). For more details on the implementation, feel free to check the commit [here](https://github.com/Midas75/AirSim/commit/1a1169b093601871b2366315be2d81b8d9a463c0).
In summary, you will need to make changes in the following three files:
- AirLib/include/common/ClockBase.hpp: `virtual void sleep_for`
- AirLib/include/common/CancelToken.hpp: `bool sleep(double secs)`
- AirLib/include/common/common_utils/ScheduledExecutor.hpp: `static void sleep_for(TTimePoint deplay_nanos)`
Please note that my fork introduces Boost as a dependency, and the build has only been tested on Windows.
Good luck!

Coinstine, 2025-04-24:
> > > AirSim adopts a spin wait similar to sleep (0) in some low-level sleep implementations, which causes threads to be repeatedly scheduled on the CPU core when multiple drones are simulated simultaneously, resulting in high kernel state consumption. When the number of simulated drones approaches the number of CPU cores/CPU threads, performance will plummet, so it can be seen that in your configuration, 80 drones correspond to your CPU96 core. The solution is to replace its internal spin implementation (approximately four times) with a plain sleep implementation and recompile AirLib for use in the Unreal project. Using this method, I can achieve 400+drone simulation based on i7-12700 in Unity scenarios.
> > 
> > 
> > hi dude, could u tell me the detail how u replace the internal spin implementation with a plain sleep one? thank u sooooo much!!
> 
> Hi, I forked AirSim and applied the implementation, which you can find [here](https://github.com/Midas75/AirSim). For more details on the implementation, feel free to check the commit [here](https://github.com/Midas75/AirSim/commit/1a1169b093601871b2366315be2d81b8d9a463c0). In summary, you will need to make changes in the following three files:
> 
> * AirLib/include/common/ClockBase.hpp: `virtual void sleep_for(TTimeDelta dt)`
> * AirLib/include/common/CancelToken.hpp: `bool sleep(double secs)`
> * AirLib/include/common/common_utils/ScheduledExecutor.hpp: `static void sleep_for(TTimePoint deplay_nanos)`
> 
> Please note that my fork introduces Boost as a dependency, and the build has only been tested on Windows. Good luck!

膜拜大佬！！！真的牛！非常感谢，这就去研究下