microsoft/AirSim #4951: AttributeError: 'MultirotorClient' object has no attribute 'getPosition'

majingchunn, 2024-05-04:
D:\anaconda3\anaconda3installation\envs\yolov5\python.exe D:\ue_installer\airlearning-ue4-master2\airsim_rl\SAC.py 
Traceback (most recent call last):
  File "D:\ue_installer\airlearning-ue4-master2\airsim_rl\SAC.py", line 676, in <module>
    sac(device=device)
  File "D:\ue_installer\airlearning-ue4-master2\airsim_rl\SAC.py", line 401, in sac
    env = AirSimEnv(need_render=False)
  File "D:\ue_installer\airlearning-ue4-master2\airsim_rl\gym_airsim\envs\AirGym.py", line 49, in __init__
    self.prev_state = self.init_state_f()
  File "D:\ue_installer\airlearning-ue4-master2\airsim_rl\gym_airsim\envs\AirGym.py", line 80, in init_state_f
    now = self.airgym.drone_pos()[:2]
  File "D:\ue_installer\airlearning-ue4-master2\airsim_rl\gym_airsim\envs\airlearningclient.py", line 100, in drone_pos
    x = self.client.getPosition().x_val
AttributeError: 'MultirotorClient' object has no attribute 'getPosition'

Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

majingchunn, 2024-05-04:
当我成功运行client.py后，我开始运行sac算法文件，报错

LeightonWang, 2024-05-09:
原版AirSim并没有这个api，这应该是AirLearning自定义的方法的相关问题，建议去那边的仓库提issue。

另外给出你具体的文件、用英文提问可能会有更多的回答。